OctoPrint/OctoPrint #701: connect options disappear

scifiaholic, 2015-01-02:
I am using the latest octopi image and this is my first try at using octoprint. My trouble is getting it to see my printer. I already know that my printer has a baudrate of 115200, and I know octoprint tries to connect with 250000. I also know that this used to be related to a problem with pyserial and python 2.6, but I have checked and I have python 2.7 on this image of octopi. The real trouble seems to be that I never have a chance to set the baudrate... the connect section of the GUI that shows both the baudrate and the port pops up then disappears instantly and completely. Its not even there as a heading that I can expand, its just gone. Sounds like a bug to me.

foosel, 2015-01-12:
Uhm, just for the record, did everything work fine once you actually logged in?

scifiaholic, 2015-01-12:
Yep :D

foosel, 2015-01-12:
In that case I'd like to close this actually. That you have to enter your login information in yourself right after creating the account in the first-setup-dialog is actually intentional, this way you have to repeat it one more time and hopefully every single password manager on earth (including your brain ;)) should then have caught on that it should be saved. I don't want to change this for this exact reason.

scifiaholic, 2015-01-12:
Well that's exactly why my suggestion at the top was geared to making it more obvious as to why the interface was completely dead.

foosel, 2015-01-12:
I guess then I didn't understand your problem at all. What exactly was it that happened? I understood you that you didn't noticed that you weren't logged in right after creating the account in the first run dialog and thought something was broken because you couldn't do anything but view (and click Log In in the upper right corner), and that once you actually logged in everythign was fine. What did I misunderstand here?

scifiaholic, 2015-01-12:
My bad, I changed that post like 3 times. Yeah, I just thought that since you want the log on event to occur, and since I didn't find it all that obvious that I still needed to log in, that perhaps you might create a separate log in page, or big notification bubble that pops up for the noobs like me.

fotis400, 2015-01-21:
You could also completely remove the "Control" tab while not logged-in, since it is empty.

I had an incidence once, trying to troubleshoot my config.yaml, just because I wasn't logged in!

foosel, 2015-01-22:
It is not empty if you have a webcam configured

foosel, 2017-01-27:
The new setup wizard in 1.3.0 now logs you in right after you set up the admin account. Let's just hope that won't lead to people forgetting their passwords directly after initial setup ;)

closed 2017-01-27 by foosel