OctoPrint/OctoPrint #702: [Request] GCode Viewer Defaults

chadwalber, 2015-01-03:
Every time I open the GCode Viewer Tab, I have to set the renderer options. Is there a way to make these options the defaults for when I load the tab?

Klikini, 2015-01-03:
Could the last-used options be set per-user in users.yaml?

chrisbcole, 2015-03-01:
I also find this frustrating.

eboston, 2015-03-01:
And how about adding an option to control how big a gcode file is before triggering the "it's too big"  warning.

Klikini, 2015-03-05:
Yes, I get that for files that I have no problem with.

asraelis, 2015-03-14:
I can not watch gcode viewer if I am using large gcodes as well. Unfortunately I can not see the gcode viewer after clicking on "OK" as well :-(. 
Anyway it's a super printing software and I am happy that I decided to use octoprint ;-)

foosel, 2015-03-23:
Duplicate of #443

closed 2015-03-23 by foosel