OctoPrint/OctoPrint #705: [Request] Manual start of a timlapse video

mlkoch, 2015-01-04:
I have not found a way to manually trigger a timelapse video.
I use a UM2 and always print using the printer's display-interface and sd-card.
Manually triggering a timlapse-video would be very helpful!

Best regards!
Michael

smsagent, 2015-01-18:
Hello,
I would like to add this feature too - while printing from SD card, only short still image of final print is made. It should be fine to have both options - timed and on Z axis change. I think, that OctoPi should read axis position while printing from SD card.

Regards!
Ondrej

foosel, 2015-01-18:
> I think, that OctoPi should read axis position while printing from SD card.

OctoPrint. OctoPi is the Raspberry Pi image. And no, it shouldn't. It would have to constantly query the data from the printer in order to achieve this and STILL would be off. That would lead to the printer firmware having to constantly answer queries (and hence risk slowing down processing) without adding real value. It just doesn't make sense to do that.

mlkoch, 2015-01-18:
I just thougt of a function to trigger the timlapse video by hand.
The trigger for the pictures needed for the video could be done by time (e.g. 120 sec.). Trigger by z-axis is not necessary.
That's all.
Best regards
Michael

foosel, 2015-02-22:
Duplicate of #452

closed 2015-02-22 by foosel