OctoPrint/OctoPrint #706: Error messages

F3000, 2015-01-06:
1. So i finally build up my rasberry pi with octopi
2. I expected, that i can read the whole error state on the left side of the Octoprint homescreen
3. Instaed i can only read half of the sentence ;(
[image]

4. Branch Version 1.1.1.1-14-g87
5. I have the RF1000 printer from conrad electronics (germany)
6. -
7. 

[image]

[image]

I hope anyone can help me and yes i love cookies ;)

foosel, 2015-01-12:
Have you tried increasing your communication timeout? => [FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#i-see-a-lot-of-communication-timeout-messages-in-the-log)

derpicknicker1, 2015-01-18:
I think its not about the error but about reading the whole error message. But the message gets turncated.
I think it was something like 
"Changing monitor state from 'printing' to 'Error: Printer requested line %d but no sufficient history is available, can't resend %lineToResend"

commit 83790fcd (2015-01-19): Do not truncate error reported from printer in UI

Closes #706
closed 2015-01-19 by foosel