OctoPrint/OctoPrint #707: Cannot login

Arengorn, 2015-01-08:
1. What were you doing?
  * I had just switched my OctoPi to devel branch and updated.
2. What did you expect to happen?
  * To type in my user/passoword and successfully login.
3. What happened instead?
  * I clicked on the Login tab, which opened normally. As I click either the username or the password textboxes, the 'remember me' checkbox or the Login button the tab closes. If I click and hold on the textboxes the tab remains open so I can type but also fails when I attemp to click on Login.
4. Branch & Commit or Version:
  * Version: 1.2.0-dev-383-g3d5fdf2 (devel branch)
5. Printer model & used firmware incl. version
  * Makerfarm Prusa i3v and latest Marlin firmware.
6. Link to octoprint.log on gist.github.com or pastebin.com:
  * Don't have it since I can't login to enable logging.
7. Link to contents of terminal tab on gist.github.com or pastebin.com:
  * [Here you go](https://gist.github.com/Arengorn/c8969af0978bcc5ba4d2)

Arengorn, 2015-01-08:
Oh yes; I forgot the pinnacle of this bug report:
## I love cookies

foosel, 2015-01-08:
Have you tried clearing your browser's cache?

Arengorn, 2015-01-08:
That worked @foosel . Simple enough solution; should have tried this before posting here. Thanks a lot  Gina.

closed 2015-01-08 by Arengorn

Atreidae, 2015-02-10:
Just did the same thing myself.
Suggest adding this to the FAQ for updating Octoprint.

Also remember you have to close the Tab after removing the cookies :(

foosel, 2015-02-10:
It IS in the FAQ!

gorbehnare, 2015-07-03:
Hello

I have the exact same issue described above on any browser (IE, Firefox, Chrome) from any computer (Two computer running Windows 8.1 Pro and Enterprise, and one Windows 7, so this is not a browser cookie issue). the issue is as soon as I click on the login username and password fields they disappear. Even if I succeed in filling up the fields by using Tab the login button does not work. Did the code issues made it's way to Version: 1.1.1-30-g4fede5a (master branch)? 

It was working fine after the initial setup, when it did create the user name and password. It seems that it is broken out of the box?

gorbehnare, 2015-07-03:
Now I have updated to Version: 1.2.2 (master branch) and still the same issue... -.-"

foosel, 2015-07-03:
https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report