OctoPrint/OctoPrint #708: logo in svg

b4zz4, 2015-01-09:


savorywatt, 2015-01-13:
@b4zz4 did you want the logo in svg format or something?

b4zz4, 2015-01-13:
El 2015-01-12 23:47, savorywatt escribió:
> @b4zz4 did you want the logo in svg format or something?

octoprint logo in svg

-- 
bazza

savorywatt, 2015-01-13:
Estoy tratando de entender si usted quiere una copia del logotipo en SVG o si desea que el logo que se muestra en el navegador web para estar SVG. No entiendo lo que quiere hacer. Siéntase libre para compartir en su idioma nativo (estoy usando Google Translate).

I'm trying to understand if you want a copy of the logo in svg or if you want the logo that shows up in the web browser to be SVG. I do not understand what you want done. Feel free to share in your native language (I'm using google translate).

b4zz4, 2015-01-13:
El 2015-01-13 01:48, savorywatt escribió:
> Estoy tratando de entender si usted quiere una copia del logotipo en
> SVG o si desea que el logo que se muestra en el navegador web para
> estar SVG. No entiendo lo que quiere hacer. Siéntase libre para
> compartir en su idioma nativo (estoy usando Google Translate).

quiero una copia del logo en svg, para poder modificarlo o hacer un 
modelo 3D basado en él.
gracias por la traducción.

-- 
bazza

savorywatt, 2015-01-15:
"I want a copy of the logo in svg , to modify or make a
3D model based on it. thanks for the translation."

Anyone have an svg copy? Repo only has PNGs and some searching did not find one.

b4zz4, 2015-01-15:
El 15/01/15 a las 00:34, savorywatt escibió:

>
> "I want a copy of the logo in svg , to modify or make a
> 3D model based on it. thanks for the translation."
>
> Anyone have an svg copy? Repo only has PNGs and some searching did not
> find one.
:(

PNG is not a project format can not be easily edited. Does the logo is not free ?, if so should be in a format project (Free )

foosel, 2015-01-15:
I have the logo in SVG. I have not published it publicly since right at the start of that project I decided that I wanted to minimize the number of people or rather corporations who would try to cash in on my work by slapping the logo on to something and then tricking people into giving them money believing they were supporting the project, or using an officially supported derivative. I'm looking into how to make the SVG public but protect the brand in the process (like e.g. Raspberry Pi, Firefox, Thunderbird, Chrome, ...) but actual development work so far had to take precedence over that - I'll try to prioritize that. Until then, every one who wants to use the logo for *personal* projects can just drop me an email (like people have done in the past).

----

Tengo el logo en el formato SVG. Pero habia decidio en el empiezo del projecto que no publicarlo a causa de minimizar el numero de las personas o mas importante las empresas que intentariá de monetizar mi trabajo y poner el logo a cualquier para venderlo, haciéndose pasar por ser oficialmente asociado con el proyecto. Estoy investigando como publicar el logo SVG pero proteger la marca (como hacerlo Raspberry Pi, Firefox, Thunderbird, Chrome, ...), pero desarollo ahora tiene precedencia - pero voy a tratar de priorizar eso. Hasta entonces, todas las personas que quieren utilizar el logo para proyectos *personales* pueden enviarme un email (es comó otras personas han haciendo esto anteriormente). (Espero que esto es comprensible, he estado aprendiendo espanol desde el abril)

closed 2015-01-15 by foosel

b4zz4, 2015-01-15:
Podrias publicar el logo con licencia de producción de pares, en otro 
proyecto para no permitir el uso comercial para empresas grandes, solo 
para cooperativas.
(es muy bueno tu español)

http://endefensadelsl.org/ppl_deed_es.html
http://p2pfoundation.net/Peer_Production_License

MiggyMan, 2017-01-26:
On the subject of people using the octoprint brand, are you aware of this ?

https://www.indiegogo.com/projects/indie-the-most-reliable-desktop-3d-printer#/

z3dev, 2017-01-26:
http://www.z3d.jp/lab/tracetool.php

foosel, 2017-01-27:
@MiggyMan Took a look, don't see anything that I'd constitute outside of fair play there ("tested with..." is stating a fact (hopefully)). Unsure about the claim of financial support they make because their name doesn't ring a bell at the moment, but that can easily be clarified (and I'm going to do that). Am I missing something?

MiggyMan, 2017-01-27:
No I just thought you should be aware of it as they seem to be pushing it as if it's "official" rather than just a value add (at least that's how I read it)

foosel, 2017-01-27:
@MiggyMan I read it as "we have also tested the machine with OctoPrint, and it works fine and you can get it shipped with the software on board". That's ok. What wouldn't be ok is "This is the official OctoPrint 3d printer". Or plastering the logo over the whole printer. Stuff like that. 

In a nutshell, my goal with being protective of the brand is not to keep people from mentioning the name or referring to it or anything, but to prevent people and companies from (intentionally or unintentionally) misleading users into thinking that what they have on hand is a product that I officially endorse, stand behind and guarantee the quality of, and/or that by buying such a product they are supporting the ongoing development when in fact they aren't.

MiggyMan, 2017-01-27:
That's fair enough, I knew this was your end goal but the way they mentioned it seemed a little "suggestive" to me so I figured it best to make you aware of it.