OctoPrint/OctoPrint #710: Invalid cross-device link on Linux

dattas, 2015-01-09:
1. What were you doing?
Uploading a .stl file on a linux filesystem which has a separate /tmp filesystem
2. What did you expect to happen?
File would show up in the file manager so that it can be edited later
3. What happened instead?
The slicing UI comes up, but if you dismiss it or go to try to slice the .stl file, it doesn't show up in the file manager. File does exist on filesystem in the correct location.
4. Branch & Commit or Version:
Version: 1.2.0-dev-383-g3d5fdf2 (devel branch)
5. Printer model & used firmware incl. version
N/A
6. Link to octoprint.log on gist.github.com or pastebin.com:
https://gist.github.com/dattas/2d5ec98b0ef936604036
7. Link to contents of terminal tab on gist.github.com or pastebin.com:
N/A

I love cookies and so I think perhaps there should be a tmp directory in the .octoprint directory instead of writing the temp files in /tmp. This should prevent most of the Cross-device link errors

Renha, 2015-01-13:
maybe shutil.move must be used instead of os.rename
os.rename should be ok for ones who had /tmp on same FS

derpicknicker1, 2015-01-18:
Pls close

closed 2015-01-18 by dattas