OctoPrint/OctoPrint #713: Server is Offline error

fotis400, 2015-01-11:
I just checked out the devel branch, but unfortunatelly it does not work for me. Although I do not get any errors, tried setup.py clean and checked requirements.txt, immediately after opening the webpage gives me a server is offline pop-up. (Although it is surelly not!)

1. What were you doing?
Accessing my octoprint server.

2. What did you expect to happen?
To fully load normally.

3. What happened instead?
It loads normally and in less than a sec, I get a Server is Offline error.

4. Branch & Commit or Version:
devel branch - commit 5a3df191a5

5. Printer model & used firmware incl. version
Not applicable.

6. Link to octoprint.log on gist.github.com or pastebin.com:
http://pastebin.com/6gmtAmeg

7. Link to contents of terminal tab on gist.github.com or pastebin.com:
Not applicable.

And regarding robots censoring their masters... "I love cookies"

foosel, 2015-01-11:
Have you tried clearing your browser cache?

fotis400, 2015-01-11:
I have tried from many browsers, also I just cleared cache in my main computer (Firefox) and the problem persists.

fotis400, 2015-01-12:
I should have done it sooner... I just checked firefox's console, and as you see dataupdater.js gets a 404 error....

http://pastebin.com/R0MCZjn2

foosel, 2015-01-12:
What reverse proxy are you running in front of OctoPrint and with what configuration?

fotis400, 2015-01-12:
how can i get this information? My (just-googled-it) knowledge on reverse proxies, says that it has to do with the server, not my browsers, correct?

Octoprint runs on Debian 3.2.63-2+deb7u2 i686

foosel, 2015-01-12:
The logs you provided indicate that you have OctoPrint running under the URL `http://192.168.1.64:5001/octoprint/`. Neither the port number `5001` is OctoPrint's default, nor is the prefix path `octoprint`. While a different port number can be configured, a different prefix path is not possible in OctoPrint out of the box since it provides it's own server anyway. So there most probably is a reverse proxy (such as Apache with mod_proxy, haproxy or nginx) in place in front of it here. How did you set it up?

fotis400, 2015-01-12:
I didn't! I just changed the port, nothing else. It is a fresh install Debian with only Octoprint installed as per instructions...

In fact I just use http://192.168.1.64:5001 to access my printer, I wasn't aware of the "/octoprint" prefix till a few minutes ago....

fotis400, 2015-01-12:
Also as I just checked, the "/octoprint" prefix apears automatically only when I checkout devel.

Using master branch and trying http://192.168.1.64:5001/octoprint/ does NOT access the server. Also everything works perfect with master branch....

foosel, 2015-01-12:
Hm, could you provide your `config.yaml` please? OctoPrint should normally only generate URLs with prefixes if instructed by either a reverse proxy in front of it or a manual configuration in the yaml file.

Also of interest would be the page source since that contains a bunch of generated URLs that could help to determine what's going wrong here, can you go to http://192.168.1.64:5001, hit Ctrl+U (or click "View Page Source" in the page's context menu), look for something that looks like this:

    <script lang="javascript">
        var BASEURL = "/";
        var API_BASEURL = BASEURL + "api/";
        var PLUGIN_BASEURL = BASEURL + "plugin/";
        var GCODE_WORKER = "static/gcodeviewer/js/Worker.js";

        var CONFIG_GCODEFILESPERPAGE = 5000;
        var CONFIG_TIMELAPSEFILESPERPAGE = 10;
        var CONFIG_LOGFILESPERPAGE = 10;
        var CONFIG_USERSPERPAGE = 10;
        var CONFIG_WEBCAM_STREAM = "http://192.168.1.3:8081/video.mjpg";
        var CONFIG_ACCESS_CONTROL = true; 
        var CONFIG_SD_SUPPORT = true; 
        var CONFIG_FIRST_RUN =  false;
        var CONFIG_TEMPERATURE_GRAPH = true; 
        var CONFIG_GCODE_SIZE_THRESHOLD = 5242880;
        var CONFIG_GCODE_MOBILE_SIZE_THRESHOLD = 1048576;

        var SOCKJS_URI = "/" + "sockjs";
        var SOCKJS_DEBUG = true; 

        var UI_API_KEY = "5C8F2DB216184C13B06C510E34810147";
        var VERSION = "1.2.0-dev-395-g4e4ad0b";
        var DISPLAY_VERSION = "1.2.0-dev-395-g4e4ad0b (devel branch)";
        var LOCALE = "en";

        var ADDITIONAL_VIEWMODELS = [];
    </script>

and paste it here?

fotis400, 2015-01-12:
You were correct. There was reverse proxy configured in my config.yaml.

I still don't understand how it get there, (I just edited the default file).... Also current master (and previous versions), seem to ignore it all the past months....

Nevermind... Thank you

closed 2015-01-12 by fotis400

foosel, 2015-01-12:
On the `master` branch that setting was called differently, so that config was not taking effect there. Glad it works now!