OctoPrint/OctoPrint #714: Cannot move axis in Control tab

Arengorn, 2015-01-12:
1. What were you doing?
  - Trying to move my printhead after changing to devel branch and updating.
2. What did you expect to happen?
  - For my printhead to move down 100mm (or any other movement that I tried).
3. What happened instead?
  - Well, nothing. I can home the Axis, I can turn fans on and off, set temps, everything but moving the axis.
4. Branch & Commit or Version:
  - Version: 1.2.0-dev-386-g5a3df19 (devel branch)
5. Printer model & used firmware incl. version
  - Makerfarm Prusa i3v running latest Marlin firmware.
6. Link to octoprint.log on gist.github.com or pastebin.com:
  - There are two files [here](https://gist.github.com/Arengorn/883c893233041f979a23), one from today and other when the problem first started.
7. Link to contents of terminal tab on gist.github.com or pastebin.com:
  - Empty, just M105 requests/responses, prooving that the commands aren't even being issued by Octoprint.

Because I love cookies I believe that this bug should be fixed so I can print my paste extruder and then make octopus shaped cookies and rule the world.

Arengorn, 2015-01-12:
[What have you tried?](whathaveyoutried.com)
**In order:**
- Restarting Octoprint
- Restarting the Pi
- Furiously clicking the buttons
- Trying every possible Axis-Length combination
- Staring at my printer with a rather angry looking face
- Updating to the latest devel release
- Sleeping to reboot myself

None of the above helped, even though I had high hopes the third and last would do the trick.

foosel, 2015-01-12:
Are there any errors in the Javascript console?

foosel, 2015-01-12:
That should hopefully be fixed now with the above commit, can you verify?

Arengorn, 2015-01-12:
Just updated and tested; it works perfectly. I'm sorry I couldn't answer to your question in time.
Thanks a lot Gina!

closed 2015-01-12 by Arengorn

foosel, 2015-01-12:
You only had like 2min to answer to that question before I realized that I had overlooked a stack trace, so no worries ;)

foosel, 2015-01-12:
Btw, I loved your "What have you tried" description (and big thanks for the link!) and if you ever get around to printing that paste extruder and create OctoCookies, please provide a picture ;)

Arengorn, 2015-01-12:
Haha, thanks; I read all issues that get posted here and none have a 'What have you tried?', I think it could be a very useful addition to the bug report guidelines. Also, I'd like to point out that all steps I listed were really done.
About the paste extruder, it's a work in progress. I got granny working on some special cookie dough and building another printer to test it all on. If I ever succeed I will for sure send you some pictures :)

commit f238ef40 (2015-01-12): Better error validation for printer profiles

Data types loaded from disk were not properly ensured to match expected types and input validation also had deficits.

Should fix #714