OctoPrint/OctoPrint #715: [Support] AttributeError: 'CommandTrigger' object has no attribute 'logger'

twproject, 2015-01-12:
I'm using Devel version of Octoprint on a OctoPi enviroment
Version: 1.2.0-dev-386-g5a3df19 (devel branch)

On config.yaml i created following example an events trigger but it don't work and report errors:

config.yaml
events:
  enabled: true
  subscriptions:
  - event: Connected
    command:
    - M115
    - M117 printer connected!
    type: gcode

log:
2015-01-12 09:10:57,066 - octoprint.events - ERROR - Got an exception while sending event Connected (Payload: {'baudrate': 250000, 'port': u'/dev/ttyACM0'}) to <bound method CommandTrigger.eventCallback of <octoprint.events.CommandTrigger object at 0x28d35b0>>
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_386_g5a3df19-py2.7.egg/octoprint/events.py", line 121, in _work
    listener(event, payload)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_386_g5a3df19-py2.7.egg/octoprint/events.py", line 282, in eventCallback
    self.executeCommand(processedCommand, commandType)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_386_g5a3df19-py2.7.egg/octoprint/events.py", line 290, in executeCommand
    self._executeGcodeCommand(command)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.0_dev_386_g5a3df19-py2.7.egg/octoprint/events.py", line 311, in _executeGcodeCommand
    self.logger.debug("Executing GCode commands: %r" % command)
AttributeError: 'CommandTrigger' object has no attribute 'logger'

foosel, 2015-01-12:
Should be fixed in `devel`

closed 2015-01-12 by foosel

twproject, 2015-01-14:
now it works fine

commit f5eef06e (2015-01-12): Fixed wrong variable name

Closes #715