OctoPrint/OctoPrint #718: Unable to disconnect

RFranta, 2015-01-14:
I tried to disconnect with my 3D printer with Marlin firmware, but when I press the Disconnect button nothing happen.
I tried it several times, I am not able to disconnect without unplugging the 3D printer cable.
I am using Olinuxino A20 (similar to RPi) and browsers Midori and Google Chrome.
I tried version OctoPrint 1.1.1-4-gc35b7a6-dirty (master branch)
2014-06-13 and also today I updated to newest version by "git pull && python setup.py install"
Firmware is Marlin about 6 months old version with megatronics board. 

Here are my log files (I just tried to connect and disconnect)

Octoprint log:
http://pastebin.com/8nTzGxEG

I love cookies

Octoprint terminal:
http://pastebin.com/2Yce22PX

foosel, 2015-01-14:
Your working copy is dirty according to the version string. Please make sure that this issue is not caused by any changes you made to the code.

Also check your browser's error console.

RFranta, 2015-01-14:
Here is the log from the updated Octoprint (it's not dirty now):
http://pastebin.com/6vhWjxDs

But problem still remain.

Here is console log:
http://pastebin.com/rDG5HhmM

There is error "Failed to load resource: the server responded with a status of 404 (NOT FOUND) http://192.168.1.103:5000/static/js/lib/jquery/jquery.min.map"
Could that be the problem?

foosel, 2015-01-15:
For whatever reason github did decide not to notify me of this via email like usual, so sorry...

The jquery.min.map is not the reason of that. Hm.. It's odd that there's neither a javascript error nor an error in the log. The disconnect button does nothing really but just remove the connection object internal, so I'm a bit at a loss what could be going wrong here, other than the printer not being actually connected and for whatever reason the UI getting confused about that. 

Did this work for your previously?

RFranta, 2015-01-15:
No it didn't work previosly in Octoprint. But connecting and disconnecting works in other softwares like Repetier, Pronterface and ReplicatorG...

Thanks for helping me.

foosel, 2015-01-15:
I'm a bit at a loss to be honest. Since it's a completely internal process I have no idea what could be causing this. 

Could you try to start OctoPrint manually from console please? Maybe that would help, might be an exception that's not being caught properly and not making its way into the log. Stopping the regular server process and starting the application manually from it's checkout folder using `python ./run --debug` (using the python command from the same python installation OctoPrint is installed under) should start up the server but not detach and hence write a lot of stuff to the console. If you then connect and try to disconnect, maybe we'll get something useful.

RFranta, 2015-01-15:
Here is the console output:
http://pastebin.com/SJa5cYYX

But it when I press Disconnect nothing change in console output.

foosel, 2015-01-15:
I have to admit, I am at a complete loss (also regarding your other issue). Never seen this, can't reproduce it at all, and if there's not even a thing on standard output, I have no idea anymore. I fear I'll have to capitulate her for now, I'm simply out of ideas, sorry.

foosel, 2017-01-27:
Two years later and no one else chimed in with the same issue, and I also couldn't reproduce anything like described above in my own tests. So it seems like that issue is restricted to your setup and nothing I can do anything about or have even a remote idea of what it could be. Sorry, but I'm closing this now as unreproducable.

closed 2017-01-27 by foosel