OctoPrint/OctoPrint #719: Gcode viewer out of the window

RFranta, 2015-01-14:
When I go to gcode viewer tab I dont see the gcode platform. I found that is somewhere out of the window and if I press the left mouse button and drag down I can finally see ithe platform, but next time when I restart my device I dont see gcode viewer again. Can I set the position of gcode platform to be lower on start?

I am using Olinuxino A20 (similar to RPi) and browsers Midori and Google Chrome.
I tried version OctoPrint 1.1.1-4-gc35b7a6-dirty (master branch)
2014-06-13 and also today I updated to newest version by "git pull && python setup.py install"
Firmware is Marlin about 6 months old version with megatronics board, but because I don't see gcode viewer even without connecting, problably doesn't matter.

Here are my log files

Octoprint log:
http://pastebin.com/8nTzGxEG

I love cookies

Octoprint terminal:
http://pastebin.com/2Yce22PX

foosel, 2015-01-14:
Your working copy is dirty according to the version string. Please make sure that this issue is not caused by any changes you made to the code.

Also check your browser's error console.

RFranta, 2015-01-14:
Here is the log from the updated Octoprint (it's not dirty now):
http://pastebin.com/6vhWjxDs

But problem still remain.

Here is console log:
http://pastebin.com/rDG5HhmM

There is error "Failed to load resource: the server responded with a status of 404 (NOT FOUND) http://192.168.1.103:5000/static/js/lib/jquery/jquery.min.map"
Could that be the problem?

foosel, 2015-01-15:
And github also didn't notify me of this reply. Grmpf :/

Ok. That jquery error is not the reason here. You are seeing the same behaviour in both midori and chrome?

How are your Renderer Options set beneath the viewer? This should be the default:

[image]

RFranta, 2015-01-15:
Yes, I have default Renderer options same as on you rscreenshot and yes,
it look the same both on Chrome, Firefox and Midori.

foosel, 2015-01-15:
Is this with or without a loaded file? If with a loaded file, can you supply the file? If without, can you try with one?

RFranta, 2015-01-15:
Its both with and without, but any file is loaded properly in gcode viewer so that shouldnt be the case.

foosel, 2015-01-15:
As said in the other ticket of you, I'm at a complete loss regarding this too. Nothing in the javascript console, can't be caused by a loaded file either, no weird options. I'm out of ideas and I fear I'll have to capitulate for now, sorry

RFranta, 2015-01-15:
Only different think is that I am using Olinuxino, not RPi. 

Is there any variable in Octoprint code that would move the platform on start in Y axis? I know its not a nice solution but it would help me. Thanks

nophead, 2015-01-15:
Actually I have seen the grid appear not in the centre of the window sometimes. I just drag it to where is should be.

irrenhaus, 2015-01-17:
I can confirm this too, only experienced it after using a fresh installation of OctoPi (like around 15 days ago). I think it was running version 1.1.1 (not exactly sure, I'm currently not using it). Checking the "Center viewport on model" checkbox fixed it each time.

If there's anything I can check let me know, I will then try to reproduce it.

Arengorn, 2015-01-18:
I can also confirm this. It used to happen every single time I uploaded and loaded a Gcode, but ever since I moved to devel (**1.2.0-dev-395-g4e4ad0b**) this has been much less recurring. At times if the Gcode file was very large, above 20Mb or so, the grid would be out of the window and it would 'stick' to the mouse making it impossible to center. Neither 'center viewport on model' nor 'zoom in on model' options would help in that, or in any case.

foosel, 2015-01-19:
I just managed to successfully reproduce this (or at least something that looks suspiciously like it) on `master`. That's at least a start I guess.

foosel, 2015-01-19:
I just pushed a change to **``master``**, can everyone affected by this please check if that solves the issue for you? If not, please provide a screenshot.

RFranta, 2015-01-19:
Now it's bed centered great! Works nice both on Olinuxino LCD display in Midori and in my PC in Chrome. Awesome. Thank you very much!

foosel, 2015-01-22:
@Arengorn @nophead @irrenhaus Can you give it another test too?

nophead, 2015-01-22:
That is worse for me because the model is no longer in the centre of the grid, presumably because my origin is 0,0, not 100,100. Previously the model was centred on the grid but the grid was not always centred. That was easy to fix by dragging it.

irrenhaus, 2015-01-22:
I will have a look this evening or tomorrow, whenever I find time for it.

foosel, 2015-01-22:
@nophead Ah, so now you are running into what @imitation described in #677 and which is tracked in #682. I guess this bug somehow covered this up for you. I'll try to look into it, on the `master` branch you still should be able to work around it by setting an offset for the extruder 1 though like described in #677 

@irrenhaus thanks!

nophead, 2015-01-22:
Yes adding an extruder offset puts it back in the centre of the grid. Does
that have any other side effect?

foosel, 2015-01-23:
Only if you run multi extruder setups. It might have some small effect on how the gcode interpreter analyses your files, but that shouldn't really affect your. So I guess not.

irrenhaus, 2015-01-23:
@foosel Fixed for me :) Thank you very much!

foosel, 2015-01-23:
Ok, closing this then. 

@nophead I hope you can live with the offset workaround for a while until I implement #682

closed 2015-01-23 by foosel

nophead, 2015-01-24:
Yes it is working fine for me and since updating the gcode viewer always
seems to work. Previously it only worked occasionally.

Regarding the first extruder offset. I think it always needs a configurable
setting because some deltas have three nozzle arranged radially, so they
are all offset relative to the centre of the end effector.