OctoPrint/OctoPrint #720: [Support] /devel translations did not work

derpicknicker1, 2015-01-15:
I wanted to enable translations in /devel branch. Started by using _/?l10n=de_ but nothing happend.
Then i did the following steps:
```bash
python setup.py babel_refresh
python setup.py babel_compile
python setup.py install
```
Again nothing happend. So i started dev mode in chrome and the console told me, that _/static/js/i10n/None.js_ was not found. I set a breakpoint to see if babel is initialized correctly. When i break in Line 4 in _/static/js/app/main.js_ i can't see a _window[BABEL_TO_LOAD_de]_ object. So it seems like babel cant refresh it's cacalog list in `/server/__init__.py` where g.locale is set (Line 89). So the _de.js_ file can't be included in _/templates/index.jinja2_ at Line 212 (_g.locale_ seems to be "None"). Thereby in Line 68 _LOCALE_ also didn't get set correctly.

So it seems like babel did not find the language files. I looked at last build log. the language _*.js_ files where copied within build. the _*.po_ files not. Short look in setup files let me do this stepps:

in _MANIFEST.in_ i added:
```bash
recursive-include src/octoprint/translations *
```

in _setup.py_ i edited Line 286 to this:
```python
package_data = {"octoprint": package_data_dirs('src/octoprint', ['static', 'templates', 'plugins', 'translations'])}
```

Then i recompiled babel files and build octoprint.

Now it seems to work. In break mode i can see _window[LOCALE]_ set to 'de' and _window[BABEL_TO_LOAD_de]_ holding a catalog.
Sadly nearly nothing is translated. The only non-english words i found are in the file browser.The "Uploaded:" time value is translated. Nothing else. See Picture[1].
Seems like babel cant find/open/load its translation files correctly.

Please help.

System: Beaglebone Black, "thing image"[2] running Amströng Linux 2014.06, Kernel 3.14.19, OctoVersion: 1.2.0-dev-404-gb0805a9-dirty (devel branch)

Octoprint Log didn't say anytihing. 
I think Printermodel is not needed here, but i only have a replicape with nothing connected yet.

[1]
[image]

[2] http://wiki.thing-printer.com

foosel, 2015-01-15:
Wow, that is pretty embarrassing, thanks for reporting this! That's been broken now since initial release of i18n and either nobody noticed or didn't bother to tell me - and I of course didn't notice myself since I usually run from my development folder. Jikes.

In any case, I've just fixed the problem of the translation files not being correctly included in the installation (see above commit with typo - yay). I tried on a fresh virtual environment and the files were installed successfully. However, they also work just fine for me when switching to Deutsch. I could imagine that you have a caching issue there since the highlighted string gets created on the fly from within Javascript whereas the rest of the page gets created from the templating code. Could you try a shift-reload in your browser and see if this fixes things for you?

derpicknicker1, 2015-01-15:
Shift-reload does not help. Seems like only the _*.js_ files get loaded or the _*.po_ files are broken or something like that.

derpicknicker1, 2015-01-15:
Chrome console also reports:
```bash
 GET http://192.168.1.2:5000/static/js/lib/underscore-min.map 404 (NOT FOUND)
```

But i don't think this is causing the translation issue.

foosel, 2015-01-15:
Nope, it isn't. Odd.. You are sure that you did not forget the ``python setup.py install`` again after fixing setup.py? Just to make sure, please try first uninstalling OctoPrint: ``pip uninstall OctoPrint`` then reinstalling it ``python setup.py install`` (make sure to use the correct python and pip binaries, if you have the OctoPrint instance installed into a virtual env, you need to use that). I agree that it sounds like it's completely ignoring anything that is in the po files while happily using what in the javascript file, but the po files should have been copied to your python's site-packages dir following the install command (you could also check that), and that did also work just fine when I tried a fresh checkout of devel in a fresh virtual environment just now.

derpicknicker1, 2015-01-15:
I uninstalled and re-installed it. But nothing changes. I also opened the page in firefox - what i have never done before - to see if it's a caching issue. No luck.
At the moment i can't understand why it's working for you and not for me.

foosel, 2015-01-16:
Can you provide the exact commands you are using for installing? Maybe that way I can reproduce the behaviour.

Here's what I did:

```
git clone -b devel https://github.com/foosel/OctoPrint.git
cd OctoPrint
virtualenv myenv
./myenv/bin/python setup.py install
./myenv/bin/octoprint -b myconfig
```

[image]

derpicknicker1, 2015-01-16:
When i do it your way on ubuntu it runs.
On BBB i'm doing this:
```bash
git clone http://[...]OctoPrint.git
cd ~/OctoPrint
python setup.py install
./run
```

Now i did it in virtualenv like you did and it works. I'm not very familar with python (and linux) to figure out why, but now everything seems to be fine :)

foosel, 2015-01-16:
I'd expect that you'd need super user rights to write to the regular python installation (which is probably located somewhere under ``/usr/``).

So maybe that's what went wrong and ``sudo python setup.py install`` would have been enough.

But the virtual env approach is cleaner anyways.

Ok if we close this now?

derpicknicker1, 2015-01-17:
Yes. It can be closed. sudo dit the trick. Now i have to understand why (as i said: new to Linux and Python :smile:)

closed 2015-01-17 by foosel