OctoPrint/OctoPrint #722: [Request] Adding for russian translation to devel

Renha, 2015-01-16:
Hello! I have just finished translating into Russian at transifex. It would be great to see it at devel branch (i'm not sure how translation must be added, that's reason why request, not pull_request, sorry)

foosel, 2015-01-16:
There are a bunch of strings that are too long:

[image]

[image]

Any chance at all those could be shortened so they have a chance to fit the available space better? I have absolutely no knowledge of Russian at all, so sorry if this is an impossible request, but if this was possible, it would be great if you could take a look at it :)

If you want to take a look yourself: Just setup the transifex client as described [here](https://github.com/foosel/OctoPrint/wiki/Translating-OctoPrint#transifex-integration), then retrieve the translation from the server with ``tx pull -l ru`` and compile the files for use by OctoPrint with ``python setup.py babel_compile`` (make sure you are using the same python version your OctoPrint is installed under). Restart your server. You can test your translation by just appending ``?l10n=ru`` to the URL :)

Renha, 2015-01-16:
Thank you for screenshots and instruction! I'll try to fix that soon, at first look it's possible

Renha, 2015-01-17:
i'm sorry, in Russian "Cancel" in dialogs and "Cancel" to stop printing is different words, is it possible to make different translation strings for them?

At settings page of Access Control, couldn't table be little adaptive?

Renha, 2015-01-26:
Now I think Russian translation is good enough for adding, please, check if all fit ok (I recommend checking with changes from pull request #742 applied).
"Cancel" issue isn't too big, it's ok as it is now. I'll open separate request-issue for it, so if language would be added, that request-issue could be closed

foosel, 2017-01-27:
I'm not going to be able to ship translations of languages that I don't understand myself directly with OctoPrint since I cannot guarantee their quality. That's why OctoPrint supports [language packs](https://github.com/foosel/OctoPrint/wiki/Translating-OctoPrint#creating-a-language-pack) (see also [here](https://github.com/foosel/OctoPrint/wiki#language-packs)) - and why I still need to setup a proper language pack repository.

closed 2017-01-27 by foosel