OctoPrint/OctoPrint #724: [BUG] Unable to delete timelapse from UI when contains parenthesis

jneilliii, 2015-01-17:
1. What were you doing?
Trying to delete timelapse mpg within web UI.
2. What did you expect to happen?
For the file to delete and remove from list in UI.
3. What happened instead?
File remained on filesystem and in web UI.
4. Branch & Commit or Version:
Version: 1.2.0-dev-404-gb0805a9 (devel branch)
5. Printer model & used firmware incl. version
N/A
6. Link to octoprint.log on gist.github.com or pastebin.com:
No log entries added to octoprint.log
7. Link to contents of terminal tab on gist.github.com or pastebin.com:
N/A

After investigating the issue it appears to be related to parenthesis in the filename.  If there aren't any parenthesis in the filename the operation works as expected.  Running dev branch on raspbian OS wheezy version 7.

I love cookies

jneilliii, 2015-01-17:
Just verified that after renaming the file to exclude parenthesis via the console the file deleted as expected.

jneilliii, 2015-01-17:
Issue does not exist on the filemanager on the main front page where the stl and gcode files live.

derpicknicker1, 2015-01-18:
I can confirm this bug. I tested it with my Version: 1.2.0-dev-405-gf52afb8-dirty (devel branch).
It seems like _werkzeug.util.secure_filename_ causes this.
You find it in `server/api/timelapse.py` on line 53:

```python
secure = os.path.join(settings().getBaseFolder("timelapse"), secure_filename(filename))
```
Updating the line to 

```python
secure = os.path.join(settings().getBaseFolder("timelapse"), filename)
```
will make it work. But i'm not sure if it's the best idea to leave out the _secure_filename()_. Otherwise, the filenames for timelapses are taken from the _*.gcode_ files. So i think they will allready be safe?!

foosel, 2015-01-19:
They will, but input validation is still necessary here. I'll fix it to properly check that no directory traversal happens here, but that should be enough.

foosel, 2015-01-19:
Fixed in `devel`

commit 14612873 (2015-01-19): Fixed timelapse deletion

Was broken after switching to new file name sanitizing with the new file management layer, still stripped everything but ascii from the filename although timelapse file names may actually contain more than that character set since the gcode file names they are based off may too.

Closes #724
closed 2015-01-19 by foosel