OctoPrint/OctoPrint #725: Print time left wrong (but estimated print time ok)

yavinmaster, 2015-01-17:
The "print time left" and the "printed" statuses are incorrect when multiples of a single object are printed.  Interestingly, the "estimated print time" is about right.  I am using slic3r 1.2.5 experimental to pipe the gcode file directly to octoprint (in case this has any influence).

foosel, 2015-01-18:
Print time left in master is still a linear interpolation based on print time and completion percentage in master, which for a number of reasons can never be accurate. In devel the average of former print times of the same object (if available) is also taken into account to mitigate this. If this average is not available, estimated print time is used until calculated print time gets stable enough. It won't get better than this, it's a hard problem.

If you happen to be interested into the background of this, take a look [here](https://plus.google.com/106003970953341660077/posts/Morgr367ge7)

yavinmaster, 2015-01-18:
The link appears to be talking more about estimated print time, not print
time left which was my issue.  The print time estimate said about 45
minutes, but print time left started to count down from about 25 minutes so
at the end of the print, print time remaining was about -20 minutes.....so
the original estimate wasn't bad.  It was a 1.1MB file and the print
progress bar only went up to about 480k so this was also wrong.

foosel, 2015-01-19:
"Print time left" is derived from a live estimation of the total print time, which the linked post was talking about. This is not to be confused with what's written under "estimated print time" in the `master` branch (and got renamed to "Approx. Total Print Time" in `devel`), and which so far did *not* get taken into account for what's displayed under "Print Time Left" since it's not guarateed to be available.

The file size that is displayed and used for the progress display is the file sized of the file on the file system - so if that said 480K it was 480K. Are you sure it didn't get truncated?

yavinmaster, 2015-01-19:
Regarding the file size.  It displayed 1.1MB in the file list in Octoprint
and the whole model printed correctly so I don't think it could have been
truncated but the progress bar only showed up to 480K (or there abouts).
If the Print time left is using the file size progress for the time
progress, this could be the cause of the problem.  Would it help if I sent
you the gcode file?  I don't have it (as I piped it directly from Slic3r)
but it wouldn't take long to recreate and retest.

foosel, 2015-01-19:
Ah, ok. I understood it that the total file size displayed above the progress bar showed 480K, not that the progress bar only got updated to around that. Sending the file might indeed help. If you have not deleted it from your OctoPrint instance yet, you can just fetch it from there too, should be downloadable via ``http://<your octoprint instance>/downloads/files/local/<yourfile>`` (I hope I got this correct from the top of my head...) or straight from your upload folder at `~/.octoprint/uploads` on the OctoPrint filesystem.

yavinmaster, 2015-01-19:
Tried to reproduce the problem tonight (as I had delete the original g
code) but all is working as I would have expected.  If I see it happen
again I will reopen a case and ensure the gcode is saved.

foosel, 2015-01-19:
Alright, closing it for now then.

closed 2015-01-19 by foosel

stefi01, 2016-03-29:
mine was working until i updated to the newest version, now what happens is that it is about 25 to 30 minuites out, a few seconds left then the time disapears and its still around 25 - 30 min left, finishes the print ok but the timing is out, going to revert to the older version.