OctoPrint/OctoPrint #726: [Bug] /devel babel refresh in setup.py does not work

derpicknicker1, 2015-01-17:
While i was trying to refresh the translation files i found some issues in the automated refresh process.

The command to update the cataloges gets _babel.cfg_ as input file. This should be the _messages.pot_ file. so change Line 197 in _setup.py_ to

```python
self.babel_update_messages.input_file = I18N_POT_FILE
```

Now refresh works and _*.po_ files are updated correctly. If you did a babel_refresh before, you have to revert the _message.po_ files to the originals from branch. By using _babel.cfg_ for updating the _*.po_ files they get messed up. For example you can use the following command.

```git
git checkout -- message.po
```

An other issue is, that the _plugins_ directory will not be searched for messages in _*.js_ and _*.jinja2_ files. You have to add a few lines to _babel.cfg_ to get them included.

```python
[jinja2: src/octoprint/plugins/**.jinja2]
extensions=jinja2.ext.autoescape, jinja2.ext.with_

[javascript: src/octoprint/plugins/**.js]
extract_messages = gettext, ngettext
```

Now `install setup.py babel_refresh` should work fine.

derpicknicker1, 2015-01-17:
Version: Version: 1.2.0-dev-405-gf52afb8-dirty (devel branch)

Printer: Replicape

Firmware: redeem

Terminal Log: N/A

octoprint.log: N/A

I love cookies!!!

foosel, 2015-01-18:
Fixed in `devel`.

And big thanks again for that full blown analysis and suggested fix. Have you thought about actually turning stuff like this into pull requests so that you get properly credited in the commit log (trying to do my best here with mentioning).

commit 039285ff (2015-01-18): Fixed babel_refresh command as suggested in #726

Thanks to @derpicknicker1 for report and suggested fix.

 Closes #726
closed 2015-01-18 by foosel