OctoPrint/OctoPrint #727: PSEye Camera Service at Startup

alextac, 2015-01-17:
Hi, I'm trying to get my webcam to work with the Octoprint service on a Beaglebone Black but am running into a small issue. Everything boots up fine and the camera works for a few seconds (indicated by the red light) and then shuts off. The weird thing is, if I then ssh to it and run the service manually it will work no problem what am I doing wrong? Also, how would I go about adding a start webcam and stop webcam button in the control tab of octoprint? I know you can change config.yaml to add custom controls but can't seem to find one. Thanks!

After running systemctl status webcam.service:
webcam.service - Run WebCam
          Loaded: loaded (/etc/systemd/system/webcam.service; enabled)
          Active: failed (Result: exit-code) since Sat 2000-01-01 00:00:18 UTC; 13min ago
         Process: 130 ExecStart=/home/octo/webcam.sh (code=exited, status=1/FAILURE)
          CGroup: name=systemd:/system/webcam.service

After running journalctl _PID=130
Jan 01 00:00:08 beaglebone webcam.sh[130]: MJPG Streamer Version: svn rev: 3:172
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: Using V4L2 device.: /dev/video0
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: Desired Resolution: 320 x 240
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: Frames Per Second.: 30
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: Format............: YUV
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: JPEG Quality......: 80
Jan 01 00:00:08 beaglebone webcam.sh[130]: ERROR opening V4L interface: No such file or directory
Jan 01 00:00:08 beaglebone webcam.sh[130]: Init v4L2 failed !! exit fatal
Jan 01 00:00:08 beaglebone webcam.sh[130]: i: init_VideoIn failed
-- Reboot --
Jan 01 00:00:09 beaglebone mpd[130]: listen: bind to '0.0.0.0:6600' failed: Address already in use (continuing anyway, because binding to '[::]:6600' succeeded)
Jan 01 00:00:09 beaglebone mpd[130]: output: No "audio_output" defined in config file
Jan 01 00:00:09 beaglebone mpd[130]: output: Attempt to detect audio output device
Jan 01 00:00:09 beaglebone mpd[130]: output: Attempting to detect a alsa audio device
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib confmisc.c:768:(parse_card) cannot find card '0'
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib confmisc.c:1251:(snd_func_refer) error evaluating name
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib conf.c:4241:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib conf.c:4720:(snd_config_expand) Evaluate error: No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: ALSA lib pcm.c:2217:(snd_pcm_open_noupdate) Unknown PCM default
Jan 01 00:00:09 beaglebone mpd[130]: alsa: Error opening default ALSA device: No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: output: Attempting to detect a oss audio device
Jan 01 00:00:09 beaglebone mpd[130]: oss: Error opening OSS device "/dev/dsp": Permission denied
Jan 01 00:00:09 beaglebone mpd[130]: oss: Error opening OSS device "/dev/sound/dsp": No such file or directory
Jan 01 00:00:09 beaglebone mpd[130]: output: Attempting to detect a pulse audio device
Jan 01 00:00:09 beaglebone mpd[130]: Assertion 'm' failed at pulse/thread-mainloop.c:232, function pa_threaded_mainloop_get_api(). Aborting.
-- Reboot --

webcam.service file:
[Unit]
 Description=Run WebCam
 After=syslog.target network.target

 [Service]
 Type=simple
 ExecStart=/home/octo/webcam.sh
 User=octo

 [Install]
 WantedBy=multi-user.target

webcam.sh file: [sleep added to see the effect  on the cam after I power up, will remove]
#!/bin/sh -
sleep 5
/home/octo/mjpg-streamer-code/mjpg-streamer/mjpg_streamer -i "/home/octo/mjpg-streamer-code/mjpg-streamer/input_uvc.so -y -d /dev/video0 -f 30 -r QVGA -n" -o "/home/octo/mjpg-streamer-code/mjpg-streamer/output_http.so -w /home/octo/mjpg-streamer-code/mjpg-streamer/www -c Username:Apassword"

alextac, 2015-01-17:
Okay, if that info is really necessary.

Version: 1.1.1-18-g868dc6d (master branch)

Printer: Davinci 1.0

Firmware: 1.15-01-02_1.1

Terminal Log: http://pastebin.com/fRDUUKxV

octoprint.log: Can't locate. Same issue as config.yaml

I love cookies?

foosel, 2015-01-18:
To be honest, I'm don't see how this report relates to OctoPrint.

Regarding your problem of finding ``config.yaml`` - it should be located at ``~/.octoprint`` of the user your OctoPrint server is run under (as can be found [on the wiki page describing OctoPrint's configuration](https://github.com/foosel/OctoPrint/wiki/Configuration)), and if it isn't, you can just create it - I added an according note to the wiki entry just now. 

About your webcam issue, I can't help you here, you'll probably get better help from some beaglebone centric mailing list or forum. My first instinct though would be that either this *User* line here:

```
[Service]
Type=simple
ExecStart=/home/octo/webcam.sh
User=octo
```

is your problem and your ``octo`` user simply lacks the necessary permissions to access the webcam, or that your webcam doesn't properly register with the video4lan subsystem right at boot. But that's only a shot in the dark.

closed 2015-01-18 by foosel