OctoPrint/OctoPrint #730: Having some trouble connecting?

Shad0wK1ng2003, 2015-01-19:
So, a few weeks ago I finished building my prusa i3 rework: that uses an arduino mega 2560, a RAMPS 1.4, and it's running marlin. I purchased a raspberry pi to run octopi and get rid of the designated computer. Now, I successfully configures octopi, however when I finally go to connect the printer, It doesn't work. Now, from my understanding, Octopi cannot support the 250000 baud rate. Unfortunately, I am unable to change the program, due to that fact that the computer with the original program crashed. If anyone has some insight unto how I would get this to work, and and all advice would be greatly appreciated.
Thanks in advance.

foosel, 2015-01-19:
OctoPi supports 250000 baud.

tiagojanz, 2015-01-20:
You can change the baudrate in the printer using the LCD.

Just go to:

Configuration->General->Baudrate  

Don't forget to save EEPROM.

I have a Rework too and its connecting with 250000 through a Octopi.

foosel, 2015-01-23:
If I'm not mistaken you also asked the same question on the mailing list and in the G+ community. One thread really is enough.

closed 2015-01-23 by foosel