OctoPrint/OctoPrint #731: [Request] Allow GET for printhead position

stv0g, 2015-01-19:
The devel version of the /printer/printhead REST-API only allows jogging the printhead in xyz coordinates and homing.

I'm currently working on a project where the current position is required for calibration purposes.
Therefore it would be really nice to have a way to query for the current position.
Maybe with a GET request instead of POST

foosel, 2015-01-19:
The problem is that OctoPrint doesn't know the print head position, it has to query it from the printer. That in turn is an asynchronous process which would block for a while on the GET request until the information has been successfully retrieved from the printer. Retrieving said information necessitates regular expression matching at the moment, since no command life cycle tracking (as in sent - received) is possible with the current communication layer. It IS possible with a) the ``commRefactoring`` branch and b) a fairly new Marlin build including [this commit](https://github.com/MarlinFirmware/Marlin/commit/acc0e7527914948656ccabba35f7faedc94ef885) or another Firmware which acknowledges commands in the same order they were received in.

Therefore this is something I'd rather add once we have proper lifecycle tracking as being worked on in ``commRefactoring`` (currently on hold due to other stuff that took priority while the above commit for Marlin was not yet merged), because everything else will just be a dirty hack.

That being said, such a dirty hack could consist of the same mechanism that's currently used for feedback controls, so maybe this helps you out in some way in the meantime.

foosel, 2017-01-27:
Proper lifecycle tracking as mentioned above will probably be something that will never be possible, at least not with associated output, simply due to how firmware works today.

There's still this annoying Marlin bug spread throughout the older forks all across the vendor versions leading to wrong acknowledgement order, and there's also Repetier Firmware acknowledging commands before sending their output, which makes definitive output tracking hard if not impossible due to not knowing when a command's output is done when none was directly sent after it (e.g. when idle).

Sending off an `M114` and actively waiting for the result matched through a regex is also something that won't be possible due to the asynchronous nature of the printer communication - it would block the API too long.

closed 2017-01-27 by foosel

FlorianJa, 2019-03-19:
Are there any news on this?

marcboivin, 2019-03-19:
See @foosel. Sadly not going to happen. Firmwares are unreliable on this either because of bugs or interpolation techniques. 

Typical low to mid-range hardware do not have the required capability to report on the actual position. You would need something like the TMC2130 drivers in order to have enough feedback to make this even remotely possible IMHO.