OctoPrint/OctoPrint #733: [QUESTION] Are eventhook commands run asynchronously?

jneilliii, 2015-01-20:
I have started using event hooks to turn on/off mjpg_streamer and was curious to know if the command list for an event is run asynchronously or not?  It seems that it may be since after adding the command to my printdone event subscription my email me command line has broken.
    events:
      enabled: true
      subscriptions:
      - command:
        - 'youtube-upload --email="jneilliii@gmail.com" --password="rfijhyomtdnceqce" --title="Timelapse: {gcode}" --description="Timelapse of {gcode}, printed and recorded via OctoPrint" --category="Tech" --keywords=OctoPrint --private "{movie}"'
        event: MovieDone
        type: system
      - command:
        - curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of {file} finished" /tmp/printDone.jpg jneilliii@gmail.com
        - sudo service livestream.sh stop
        event: PrintDone
        type: system

foosel, 2015-01-21:
They are run asynchronously within OctoPrint, but all commands from an event hook are executed in the order they are specified in. So your mail command not working anymore must have some other reason. Anything in the log?

jneilliii, 2015-01-21:
Running through a test now to see if anything is put in the log.  I also modified the mpack command to write it's output to a file by adding "> /tmp/mail.log" to the end of the command.

The work around for me was to append it to the email command using && similar to the way the recipe is showing curl and mpack running on the same command line.  This basically makes it synchronous via linux if I'm not mistaken and I did receive the email and mjpg_streamer daemon did stop as expected.  

Will update after my test run.

jneilliii, 2015-01-21:
So the only thing that shows in the log is the following.
````
2015-01-21 19:32:42,038 - octoprint.events - INFO - Executing system command: sudo service livestream.sh start
2015-01-21 19:56:07,646 - octoprint.events - INFO - Executing system command: curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of /home/pi/.octoprint/uploads/Cube.gco finished" /tmp/printDone.jpg jneilliii@gmail.com > /tmp/mail.log
2015-01-21 19:56:07,687 - octoprint.events - INFO - Executing system command: sudo service livestream.sh stop
````
and nothing seemed to be piped into the mail.log file and the image was not captured, so not really sure why it's breaking.

I obviously have a workaround for now, so no big rush, but something to consider if other people start working with event hooks and also have issues.  The below event hook works.
````
    events:
      enabled: true
      subscriptions:
      - command:
        - curl -o /tmp/printDone.jpg "http://localhost:8080/?action=snapshot" && mpack -s "Print of {file} finished" /tmp/printDone.jpg <email address removed> && sudo service livestream.sh stop
        event: PrintDone
        type: system
````

foosel, 2015-01-22:
That stuff being present in the log indicates that OctoPrint handed it off to the underlying OS in the exact order as specified, and that there also were no errors while doing that. The sub process is created synchronously, so there should be no way that the "livestream stop" command could take over the mail command. I don't understand this.

foosel, 2015-01-23:
Marked this as unreproduced bug

foosel, 2015-01-23:
Wrong button, sorry

foosel, 2017-02-23:
I couldn't believe it myself, but I finally found out why that was happening and I feel a bit stupid now. Turned out that while I was in fact executing the commands in the correct order, I was not waiting for them to finish before the next was triggered (thanks to misunderstanding how Python's `subprocess.Popen` works). And that of course allowed one command to overtake the next and by that produce problems like those you encountered, making it seem like commands were executed in the wrong order, when in fact they weren't but also didn't wait for each other.

This is now **finally** solved on `maintenance` via the above commit, will soon be merged to `devel` and will also be part of the 1.3.2 release.

foosel, 2017-03-16:
1.3.2 was just released.

closed 2017-03-16 by foosel