OctoPrint/OctoPrint #734: Server goes down when uploading large gcode files

aerophilic, 2015-01-21:
Tried uploading a GCODE file of about 75MB, and the server crashed hard. Not sure if others have seen this, but I deliberately did an update/re-install to ensure it wasn't recently fixed.

derpicknicker1, 2015-01-21:
I successfully uploaded a 86MB gcode file. 
Version: 1.2.0-dev-423-g039285f-dirty (devel branch)
System: BeagleBone Black, Linux Angström 2014.06, Kernel 3.14.19, Upload to mounted microSD

foosel, 2015-01-21:
Fixed in `devel` back in August, not yet backported since quite an invasive change which I did not want to port to the stable branch without it getting some testing first.

That being said: Duplicate of #455 and also severely lacking any information at all to be able to further debug this (like a version for example).

closed 2015-01-21 by foosel