OctoPrint/OctoPrint #735: Autoscroll always on

braddo99, 2015-01-21:
There is an autoscroll toggle button on the terminal window which does not appear to do anything. Checked or unchecked my console scrolls commands continuously. This makes it quite difficult to copy paste out of the console, which I find helpful to do when calibrating my bot. As soon as a scroll happens, the selected content becomes unselected and hence uncopyable.

I'm running on raspberry pi (octopi) and client is Chrome on Windows7

derpicknicker1, 2015-01-21:
I think it works like it should. If autoscroll is enabled, the terminal will always jump to the last line on receiving a message. If Autoscroll is disabled, it will not jump to last line. But because there will be new lines on stack, the lines will move.

braddo99, 2015-01-21:
Mmm, well if the lines move at all then one is unable to accurately select content to copy from the terminal stack. I see what you are saying, but if you are already at the last line, the behavior would be identical with or without autoscroll right? Then, if in autoscroll off, you manually scroll back, the lines will continue to move at the same rate.  I really don't see any positives of the current behavior but do see negatives.

Renha, 2015-01-22:
@derpicknicker1, you are right about enabled Autoscroll, but with Autoscroll disabled it _will_ jump, just not to last line. It is not logical, position must be constant from beginning of terminal, not from end, user doesn't care about stacks or registers etc.

foosel, 2015-02-24:
Implemented in `devel`. And now please someone explain to me what else I have to do to get you people to read and adhere to the bloody contribution and bug report guidelines...

commit e9623fdc (2015-02-24): Improved behaviour of terminal window

  * Disabling autoscrolling now also stops cutting of the log while it's enabled, effectively preventing log lines from being modified at all
  * Applying filters displays "[...]" where lines where removed
  * Added a link to scroll to the end of the terminal log (useful for when autoscroll is disabled)
  * Added a link to select all current contents of the terminal log for easy copy-pasting
  * Added a display of how many lines are displayed, how many are filtered and how many are available in total

Closes #735
closed 2015-02-24 by foosel