OctoPrint/OctoPrint #736: Feature Request: Effector position feedback on Control tab

braddo99, 2015-01-21:
I find it strange that the control tab doesn't show the actual tool position. I would like to have a readout of X Y and Z to make it easier to make adjustments like calibration.

foosel, 2015-01-21:
Just added the answer to this [to the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#why-doesnt-octoprint-show-me-the-current-hotend-position-anywhere)

closed 2015-01-21 by foosel

braddo99, 2015-01-21:
Thanks for the response on the FAQ, and what you said makes sense. However, I should have been more specific in this issue post - I don't want position feedback while the bot is printing, only moves issued by OP from the control tab after the completion of the requested move. I can check into the custom command, but imo if the up/down etc buttons also issued M114 after their intended sent gcode and there was a readout  on the control tab that would be a nice improvement.

braddo99, 2015-01-21:
Just checked and it appears that you warn in the creating custom commands documentation that they are always listening so, for the reasons you already suggested it's not a good solution for the above. Might be appropriate to reopen this ticket since it's not really addressed by your FAQ. Also I can see a similar request just arrived [#731](https://github.com/foosel/OctoPrint/issues/731) - although it's asking about getting position via GET, I can see that it is also in the context of doing calibration.