OctoPrint/OctoPrint #743: Problems with remote access (over the internet)

Carbonwasp, 2015-01-26:
Hi - I've been using Octopi for a few months on Raspberry Pi and get on really well with it. I have set up dynamic DNS to access Octopi away from my home network and it has been working well for a couple of months.
Last week I updated to the latest version of Octopi (version 27/10/14) by downloading the image and writing it to the SD card in the usual way (I now realise there's a much quicker way to update!). Once I had set the Octopi up on my wifi network (using the same wifi dongle as before) the printer worked well locally. I have no problems accessing Octopi on my local network using the local IP address (which is reserved). However, whenever I try to access Octopi away from my local network, i.e. over the internet, I get redirected from "octopi.mywebsite" (not my real address) to "octopi.mywebsite/UI" and am told the page is: "Not Found, The requested URL was not found on the server"
It seems to me (I am certainly no expert) that the local IP address of my Pi is reserved correctly on my router and that the dynamic DNS is working properly to point my web address to the router. The only change I have made is updating to the newer version of Octopi - is there some added security in the newer version to prevent/limit internet access? Or is there a setting I need to change on the Pi to make this work?
Thanks in advance for any help. Also thanks for the software, it's a fantastic interface.

Carbonwasp, 2015-01-26:
I have read the FAQ. (Sorry)
I'm using:
- Marlin firmware
- Firefox browser
- Octopi Version: 1.1.1-24-g2de863c (master branch)

Thanks

foosel, 2015-01-26:
Sorry, but you are at the wrong location (I just updated the "How to file a bug report" guide to help in this regard). This is the bug tracker of the OctoPrint software (the 3d print server and corresponding web interface), you are running more into a problem with OctoPi (the preconfigured Raspberry Pi image, which among other things includes OctoPrint). The version number you are reporting there is btw the version of the installed OctoPrint version (not OctoPi).

Now, what you describe does also not sound like a problem with OctoPi itself, but rather with your network setup. I have absolutely no idea what is happening there, but maybe your ISP changed something, or your router got some weird update - it certainly doesn't sound like something that any update of OctoPi (or OctoPrint on it) could be causing. If everything works fine locally, it IS your network setup.

I'd suggest to try if you can reach other addresses within your network from without, or if that redirect happens there too, in which case you should probably try to find other people who are with the same ISP (support forum?) and look if they are experiencing similar issues.

closed 2015-01-26 by foosel