OctoPrint/OctoPrint #745: Error importing util

MYasirSheikh, 2015-01-27:
Hi,

I have recently switched to devel branch and have been trying to run the server. The point I'm stuck at is when the __init__.py script (present in src/octoprint/server/) tries to import util.
line 59: from . import util

The problem is that there is no file named util.py in the above mentioned folder.
What should I do to get that file??

Regards
Yasir

MYasirSheikh, 2015-01-27:
Also if i try to use the util.py present in the server folder on foosel OctoPrint, it removes the above error but gives me a new one

pi@raspberrypi ~ $ ./OctoPrint/run
2015-01-27 13:29:05,660 - octoprint.server - INFO - Starting OctoPrint 1.2.0-dev                                                                                        -442-g6576291-dirty (devel branch)
2015-01-27 13:29:05,674 - octoprint.plugin.core - INFO - Loading plugins from /h                                                                                        ome/pi/.octoprint/plugins, /home/pi/OctoPrint/src/octoprint/plugins and installe                                                                                        d plugin packages...
2015-01-27 13:29:06,094 - octoprint.plugins.discovery - INFO - pybonjour is not                                                                                         installed, Zeroconf Discovery won't be available
2015-01-27 13:29:06,103 - octoprint.plugin.core - INFO - Found 2 plugin(s): Disc                                                                                        overy (0.1), CuraEngine (0.1)
2015-01-27 13:29:06,119 - octoprint.plugins.cura - INFO - Path to CuraEngine has                                                                                         not been configured yet or does not exist (currently set to None), Cura will no                                                                                        t be selectable for slicing
2015-01-27 13:29:06,128 - octoprint.filemanager - INFO - Adding backlog items fr                                                                                        om local to analysis queue
2015-01-27 13:29:06,165 - octoprint.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "./OctoPrint/run", line 22, in <module>
    octoprint.main()
  File "/home/pi/OctoPrint/src/octoprint/__init__.py", line 81, in main
    octoprint.run()
  File "/home/pi/OctoPrint/src/octoprint/server/__init__.py", line 419, in run
    appSessionManager = util.flask.AppSessionManager()
AttributeError: 'module' object has no attribute 'flask'

Is that file the right one.. what else can be the issues?

foosel, 2015-01-27:
*Properly* update to the ``devel`` branch, which apparently you didn't do, because otherwise it would find [the module it is trying to import just fine](https://github.com/foosel/OctoPrint/tree/devel/src/octoprint/server/util). Make sure you really are on the ``devel`` branch, that your branch is clean (``git status`` doesn't show anything to commit). Also try ``python setup.py clean``.

A personal word though: There are [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md), and it seems like you still haven't read them although this is your second ticket where the bot tells you to please do that and although they are linked at the top of every ticket submission form. You are not encountering a bug, you don't have a feature request, and these guidelines will tell you that this bug tracker is not the right location for support requests. I tried in the past to do support via this bug tracker. It doesn't work. The cited locations for getting support should get you help faster since there are more people than just me and some people interested in the *development* of OctoPrint who might see your question and respond.

closed 2015-01-27 by foosel