OctoPrint/OctoPrint #747: language support

b4zz4, 2015-01-30:
I want to translate the program into Spanish

yo deseo traducir el programa al español. como hice con varios con cura, openscad

foosel, 2015-02-04:
Hola @b4zz4, la aplicacion ya estado traducido al espanol en su mayoria. Puedes buscar la coordinacion y traducion en [Transifex](https://www.transifex.com/projects/p/octoprint/) y los instrucciones para traducir en [esta pagina del wiki](https://github.com/foosel/OctoPrint/wiki/Translating-OctoPrint). La version espanol no aun esta merged en el branch ``devel`` porque lo tengo que review - si tu quieres me puedes ayudar y review la traduccion y test como se ve.

(Espero que me has entiendo, aun estoy aprendiendo el espanol)

closed 2015-02-04 by foosel

b4zz4, 2015-02-20:
El 04/02/15 a las 20:45, Gina Häußge escibió:

> Hola @b4zz4 <https://github.com/b4zz4>, la aplicacion ya estado
> traducido al espanol en su mayoria. Puedes buscar la coordinacion y
> traducion en Transifex [...]

necesito tener una cuenta para bajar las versiones en español, no están
en el repositorio :(
no me gusta tenes que tener una cuenta para descargar la versión de
transifex<https://www.transifex.com/>

sgregori, 2020-05-01:
Coincido contigo, vaya rollo tener que registrarse y perderse en esa web de transifex.

Aqui dejo el fichero en español.

[for_use_octoprint_messages_es.zip](https://github.com/OctoPrint/OctoPrint/files/4562954/for_use_octoprint_messages_es.zip)