OctoPrint/OctoPrint #751: Chrome 40/41 running out of memory consistantly

shanselman, 2015-02-01:
After several hours Chrome 40/41 is running out of memory consistently. I'll run it again with DevTools on.

shanselman, 2015-02-14:
Added info

darylc, 2015-11-01:
I am also experiencing this issue.

foosel, 2016-01-25:
I still haven't been able to reproduce this but it might actually have the same core reason (or be the core reason?) as #1065, which I also haven't been able to reproduce myself but for which I pushed a potential fix a while ago that so far appears to have solved the issue for a couple of people who reported back on it.

@shanselman, @darylc have you been able to reproduce this with 1.2.8?

foosel, 2016-02-15:
Any updates on this against 1.2.8 or 1.2.9?

hyperair, 2016-06-16:
Happens in 1.2.8 for me, but only on lower-RAM machines (recently got bumped down from 16GB to 8GB). I think it might be a mjpg-streamer issue, but I can't be sure.

jexoteric, 2017-03-31:
I am also having an issue with this, although it does not crash the tab, my entire computer comes to a crawl as it swaps memory in and out. OSX 10.12.3, Chrome 57, Octoprint 1.3.2. Closing the tab for octoprint brings my computer back to normal operation.

foosel, 2019-11-05:
Have not heard of further reproductions in the past 2.5y or been able to repro myself, assuming this is fixed. If it still happens with current versions, please open a new ticket, the code base has moved a lot.

closed 2019-11-05 by foosel