OctoPrint/OctoPrint #753: [Request] Sub-directories in File listing

eboston, 2015-02-01:
When working on projects that contains several parts, it would be nice to be able to upload the files to a sub-directory to keep everything related to the one project together and separate from everything else.  It would also be helpful when I have a project that I will often go back to and reprint.  Would help keep the file listing from getting cluttered.

v8surf, 2015-02-04:
While were at it with the file listing... would it be possible to include a refresh option as I have found at times that large gcode files (30-40MB) take forever and then usually fail.  
I get around this by manually loading them to the uploads folder, but then have to restart octoprint as well.

foosel, 2015-02-04:
@eboston directory support is already in the ``devel`` branch, just not reflected in the UI yet, there exists a ticket for that already though => #552 . Therefore I'm marking this as a duplicate and closing it.

@v8surf while a refresh option would technically be possible, I would instead suggest to a) check out the ``devel`` branch and test if the overhauled upload handling in that one maybe already solves your problem and b) if not, use the watch folder feature in that one, which triggers a file list refresh if it detects a new file within the watched folder.

closed 2015-02-04 by foosel