OctoPrint/OctoPrint #754: [Request] Have software version check and auto update option

foofoodog, 2015-02-01:
Wondering if it would be possible for the software to offer an auto update option to run the git pull, the setup.py clean/install and the service restart. Might only apply to me running Octopi on several devices and being lazy about updating them to get fixes. Bonus points if we could switch between master and devel branches. 

I love cookies and cookies love me.

foosel, 2015-02-01:
One step ahead: https://github.com/OctoPrint/OctoPrint-SoftwareUpdate

Not properly documented yet, and depends on the plugin system present in the devel branch, but fully functional.

closed 2015-02-01 by foosel

foofoodog, 2015-02-02:
Very cool, and it backs up my local changes if I have any as a patch, that is a nice touch.