OctoPrint/OctoPrint #755: Delegating gcode control flow to a plugin [Brainstorming]

platsch, 2015-02-03:
Hi,

I'm working on a project where we try to handle objects (smd-components) during the printing process by the printer itself. This requires feedback from a camera system to find and align the objects. To achieve this, we plan to extend the gcode files with information for the placing system, which is then supposed to control the printer interactively.
Obviously, this is not possible with the current (quite linear) way octoprint and other host software feeds gcode to the printer.
I am aware of the buffered communication with the printer firmware, the placing system has to cope with this (e.g. by using M400 commands).

I'd like to implement the "realtime" controller as a plugin, but I would need to "interrupt" the communication while loop which is not possible with the existing plugin mechanisms. Therefore I consider adapting the comm-loop and introducing a new state e.g STATE_PRINTING_EXTERNAL. Then a plugin could hook to the event bus, wait until the comm-loop sets the state to STATE_PRINTING_EXTERNAL, recieve the extended-gcode commands (maybe as event payload?), do the object manipulation and return the control flow to the comm-loop by setting the state back to STATE_PRINTING.

Is the idea comprehensible?
Any opinions?

foosel, 2015-02-04:
Just chiming in quickly to say: I've seen this, I'm going to reply properly, I'm just on a business trip at the moment and haven't yet come around to properly think about it, sorry.

platsch, 2015-02-06:
Ok, I started digging deeper into the (partly not yet documented) plugin mechanism. I get the impression that it should be possible to implement this by using hooks and events without any modification at octoprint itself. 

A first prototype should be done by next week, I'm going to report the outcome, consider this as solved for now. I leave the issue open for further hints and ideas, but feel free to close it if you prefer a small nr of open issues :)

foosel, 2015-02-06:
Sitting at the airport while waiting for a flight has the advantage to be able to at least get stuff like this tackled ;)

I was going to suggest hooks as well, but I'm still thinking about how to turn the hooks into something deterministic and "preventable", e.g., if you have multiple plugins targeting the same hook they should have a deterministic order in which they are called and depending on the hook functionality also be able to stop further procession down the chain.

From how I understand your problem, you probably could use the ``octoprint.comm.protocol.gcode`` hook (if you need to process commands embedded within the gcode file, which i'm not entirely sure is what you'd need to do). For reacting to data coming back from the printer, there's ``octoprint.comm.protocol.action`` which gets triggered through lines received from the printer that look like this: ``// action:someAction``.

Hope this helps. Also, sorry for the lack of documentation, I'm working on it, but you are really still on the bleeding edge here :)

platsch, 2015-02-06:
Hey,
yes, I started with the `octoprint.comm.protocol.gcode` and it does pretty much exactly what I'm looking for. The `octoprint.comm.protocol.action` is a good hint, thx.

I already asked myself what happens if multiple plugins would hook at the same point, but decided to wait for the documentation to be done with this point ;-)
Don't apologize for lacking documentation, it's my problem if I start using software in this early stage. You are doing a good job.
Have a good flight.

gt6796c, 2015-02-08:
Hi,
I'm looking to do something similar making OctoPrint (ditto on the good job) look after my ShapeOko2 (grbl v0.9 something.) I started with the code that someone had done a while ago off the main branch and over the past week or so got it mostly ported to the more modern plugin approach.

In the end I created two more sets of hooks. One for input (mainly needed to handle handshakes from the printer) and one for output (my grbl board doesn't like the checksums.)

I spent quite a bit of time trying to make it so that I didn't have to touch any of the baseline code, but I simply needed to be able to adjust the internal state machine of the comms to make it work. I did try to stay within the spirit of the plugin and hooking model. You can see it at https://github.com/gt6796c/OctoPrint.git on the octogrbl branch and then the accompanying plugin is at https://github.com/gt6796c/OctoPrint-grblplugin.git I didn't see how to make use of the action stuff yet. I can't control what the printer sends or can consume, so I manhandled it to work.

There are a few hacks in place mainly to avoid messing with baseline code (e.g. the plugin relies on a naming convention of the Printer Model to activate.)

I just had my first successful milling test this evening using this code. I'm about to get it deployed to my OctoPi and tomorrow I'll hook it back up to my Printrbot and make sure I've not horked that up.

platsch, 2015-02-09:
I took a look at the `octoprint.comm.protocol.action` hook and I agree with gt6796c, I have no idea how to work with it. Extending the firmware to send appropriate commands seems not to be very feasible for most cases. Maybe a general hook (filtered by a regex?) might be a better idea?

foosel, 2015-02-09:
The action thingy is currently being adapted to be included in firmwares (OctoPrint is a bit ahead of its time here), see also [here](http://reprap.org/wiki/Gcode#Replies_from_the_RepRap_machine_to_the_host_computer). Smoothieware already uses this, a couple of Marlin versions out there as well. But as I said, I'm not entirely sure if it fits your usecases.

It might indeed be an idea to allow extension of the current "feedback command" stuff by hooks. Will try to look into it.

Btw, @gt6796c, absolutely awesome! It might of interest to you that there's also a completely different communication approach on the ``commRefactoring`` branch available (work in progress, for ages now *sigh*) which will allow protocol and transport layers to be provided by plugins. So something like what you are doing would be even easier.

gt6796c, 2015-02-09:
I'd be happy to take a look at doing a more robust job for consideration. My stuff still suffers from the non-deterministic hooking. Other attempts (I think on octocnc branch) tried to layer in a bit cleaner with some refactoring to include things like sendIdleCommand instead of hard coding sending M105 (as is now) or '?' Which is what grbl would expect. I could take a look at the refactor or continue experimenting with my current path. I'd think that one more layer between comms and the "printer" would be needed. I called it "GCodeFlavor" in one pass. 
Let me know if you're interested in discussing more.

platsch, 2015-02-23:
Just want to give a quick feedback: I was able to implement everything I need by using events and the gcode hook without touching octoprint itself. The plugin is already working but not very stable yet. I'm going to publish the plugin and detailed information as soon as possible.

foosel, 2015-02-23:
@platsch that sounds awesome, thanks for the feedback!

foosel, 2015-02-23:
@gt6796c sorry, saw this just now... the refactoring differentiates between a protocol (e.g. gcode in reprap style or CNC style or some other format, not necessarily gcode) and transport layer (serial, tcp, ...). That regular sending of an M105 is also part of that protocol specification and could easily be overridden by something else.

gt6796c, 2015-02-23:
@foosel, so what is the status of that branch? Anything that I could do to help move it along?

foosel, 2015-06-04:
@platsch @gt6796c fyi, I recently had to change the `octoprint.comm.protocol.gcode` hook, or rather, replace it with a bunch of others that allow more granular control/interventions (see [here](http://docs.octoprint.org/en/devel/plugins/hooks.html#octoprint-comm-protocol-gcode-phase)). The basic gist of the changes is that it's been replaced by `octoprint.comm.protocol.gcode.queuing` and that you can also hook into the other processing phases. Sorry for the inconvenience, but I'm going to push out a release candidate for 1.2.0 hopefully early next week (with a release following after that hopefully the week after, depending on feedback) and that specific hook fell victim to my final clean ups before that. But the adaptation shouldn't be too problematic (just changing the signature and the return format slightly).

If you need to hook into what is being returned from the printer in general, one option (not my idea actually, something another plugin author came up with that I should really have remembered myself *cough*) is to implement the [PrinterCallback](http://docs.octoprint.org/en/devel/modules/printer.html#octoprint.printer.PrinterCallback) interface and register with it, you'll then get all serial data from the printer. The action stuff indeed is rather of interest for special events. The hope is that it will not be something that has to be individually patched in, but that some commands will become standard. But we'll see.

@gt6796c sorry, for whatever reason I never got a notification about your question and just saw it now :( `commRefactoring` is on a back burner until 1.3.0 or something

platsch, 2015-06-05:
Thanks for the warning. That's not a problem at all, it might even simplify things :)

platsch, 2015-06-16:
Well, I'm currently updating to the newest version. The hook works perfectly so far, but the injected `self._printer` property (http://docs.octoprint.org/en/master/plugins/concepts.html#injected-properties) seems to be removed. Is that intended? Am I missing something?

foosel, 2015-06-16:
That sounds odd, it should still be there. When are you trying to use it? Do you have some code I could take a look at?

platsch, 2015-06-17:
Ok, you are right, it was my fault. I confused the plugin objects while registering the new hook. Thx!

foosel, 2017-03-08:
@platsch doing a long overdue spring cleaning of the issue tracker... did you ever publish your plugin? Do we still need to keep this ticket open or is it obsolete now (I don't see any action points)

platsch, 2017-03-08:
@foosel we can close this issue. The problem is solved. And yes, I kind of published the plugin: https://github.com/platsch/OctoPNP

I felt that it is not an important feature for most people, since it is pretty specific for hardware with a pick and place extension for SMD components and an extruder for conductive material. Also the documentation is not fully up to date...
On the other hand it would be possible to insert all kinds of small objects (magnets, nuts, etc) with this approach, but it still requires a vacuum gripper in your printer.

Good work with octoprint, writing extensions as a plugin is extremely comfortable with this infrastructure!
But we are using it actively in our lab environment and the development is still ongoing.

closed 2017-03-08 by platsch

foosel, 2017-03-08:
@platsch I'd definitely love to see it on the [repo](http://plugins.octoprint.org/help/registering/)! It might be for a very specific use case, yes, but that holds true for a lot of things and who knows, maybe someone is looking for exactly this kind of solution right now :)

platsch, 2017-03-08:
Yepp, that's a point. I put it on my todo list!

foosel, 2017-03-08:
👍 Btw, I'd love to see that in action if I happened to be in the vicinity (where ever that might be). Possible or absolute no-go? :)

platsch, 2017-03-08:
Sure, the lab is in Hamburg, just come by anytime you like :-)
Your profile says Obertshausen, I happen to by in Darmstadt today, so we could actually meet for a beer ;-)

foosel, 2017-03-08:
@platsch sorry - that's a bit too short noticed, already have dinner plans. Definitely will keep a lab visit in mind the next time I'm in Hamburg. Missed the Miniaturwunderland when I went for 32c3, so that's still somewhat high on the bucket list and would be a welcome opportunity to combine :)

platsch, 2017-03-08:
Sure, just noticed the funny coincidence. Just drop me message, visitors are always welcome.
CCCongress is definitely enough input for one visit :)

platsch, 2017-05-19:
Just for the record: the plugin is registered to the repo now.

foosel, 2017-05-19:
And I just merged that request and [it's live](http://plugins.octoprint.org/plugins/OctoPNP/) :)