OctoPrint/OctoPrint #757: Request - Add temperature to 'state' menu

kev626, 2015-02-05:
Add a spot for temperature to the state of the printer on the sidebar.

RemeberMe, 2015-02-08:
Yes, please.  I am constantly switching back and forth to view temps.  I don't need a fancy graph, just the present temp would be awesome.

foosel, 2015-02-08:
Switch to `devel` and use the [Navbar temp plugin](https://github.com/imrahil/OctoPrint-NavbarTemp)

closed 2015-02-08 by foosel

RemeberMe, 2015-02-08:
Can't.  I'm one of the happy idiots who just used the SD image because I have no idea how to do this manually :) Hal

foosel, 2015-02-08:
Which is why [it's explained in the FAQ](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image) ;)

In any case, while I will not add this to the master branch in the future such stuff will be the job of plugins as can already be seen, so rest assured it's being worked on, only slightly different than expected ;)

RemeberMe, 2015-02-10:
Thanks Gina!  I've been just a little chicken to try this till I finish off a current project :) Hal