OctoPrint/OctoPrint #763: High CPU Usage after canceling print

mosh1, 2015-02-09:
After I cancel a print I cannot operate the controls for a while which can be frustrating. Top shows Octoprint taking 80% CPU for 20s or so which explains the control freeze up. This is on master and the latest release.

Camera timelapse is off. Wondering if this is expected or a bug?

mosh1, 2015-02-09:
I have read the FAQ.

1. What were you doing?

Canceling a print then moving printhead.

2. What did you expect to happen?

Move controls should happen as quickly as feasible after canceling.

3. What happened instead?

High CPU usage locks up Octoprint for 20-30s.

4. Branch & Commit or Version of OctoPrint:

Master 2523cf81ba24d3224447db7698e72e0e989ad17f & 1.1.1 release.

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

Printrbot Simple Metal

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Safari on OSX Yosemite.

7. Link to octoprint.log on gist.github.com or pastebin.com:

http://pastebin.com/ZVjkpibf

10. Screenshot(s) showing the problem (if applicable)

```
top - 05:48:00 up 23:51,  2 users,  load average: 1.70, 1.02, 0.74
Tasks:  74 total,   3 running,  71 sleeping,   0 stopped,   0 zombie
%Cpu(s): 75.2 us, 11.3 sy,  0.0 ni,  0.0 id,  0.0 wa,  0.0 hi, 13.5 si,  0.0 st
KiB Mem:    382840 total,   225416 used,   157424 free,    22248 buffers
KiB Swap:   102396 total,        0 used,   102396 free,   130940 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND                                                                                                                                                                        
 3066 pi        20   0 61160  21m 3652 R  75.0  5.7  61:04.50 octoprint                                                                                                                                                                      
 2377 pi        20   0  116m 2672  912 S  12.9  0.7  92:58.68 mjpg_streamer                                                                                                                                                                  
 2361 haproxy   20   0  9876 5660  656 S   5.0  1.5  22:47.38 haproxy                                                                                                                                                                        
    3 root      20   0     0    0    0 R   3.1  0.0  10:52.90 ksoftirqd/0                                                                                                                                                                    
 3320 pi        20   0  4648 1456 1028 R   1.6  0.4   0:01.97 top                                                                                                                                                                            
   30 root       1 -19     0    0    0 S   0.6  0.0   4:12.07 VCHIQ-0                                                                                                                                                                        
 3311 root      20   0     0    0    0 S   0.6  0.0   0:01.17 kworker/0:1                                                                                                                                                                    
 1658 root      20   0     0    0    0 S   0.3  0.0   1:58.49 RTW_CMD_THREAD
```

mosh1, 2015-02-11:
FWIW I switched to devel branch and while Octoprint still uses high CPU after cancelling print, the controls become active much sooner. This removes my main frustration. I'm not sure if this was a fluke and I just needed a rebuild or if devel branch fixed something related to this.

dattanchu, 2015-03-20:
I am experiencing the same issue on Printrbot Simple Metal with RPi B+. Timelapse is also disabled.

After cancelling, I have to wait more than 10 seconds before the control is resumed. This doesn't happen on Cura.

foosel, 2015-03-23:
@dattanchu master or devel? The devel branch changes command processing. My guess is the CPU usage and the controls becoming unresponsive are two different issues shadowing each other.

dattanchu, 2015-03-23:
1.1.1 master branch. I can try switching to devel to confirm that the issue is fixed.

dattanchu, 2015-03-30:
Devel branch on octopi. Issue is manageable. i.e. the lag after cancelling a job is not too long. I guess we can just wait for the next stable release of octopi.

foosel, 2015-05-09:
Should be solved by optimizations in the communication layer (proper tracking of acknowledgements and hence only sending of more lines when firmware signals readiness to receive compared to "send spamming" earlier, also a less long command queue on the host side due to that). Note that regarding printer movements there is always a certain lag due to the planning buffer on the firmware side (as in, a bunch of moves already queued up) and currently there's no way by the host to clear that buffer without completely shutting down the printer. 

The CPU usage might be explained by yet unprocessed gcode files entering the analysis phase (gcode analyser gets started right after the "Printing" state is left).

However, since I'm not completely sure this is really satisfactory, feedback from you, @mosh1 , would be helpful to determine if `devel` indeed fixes things (within above limits) or not.

mosh1, 2015-05-10:
I haven't had any issues with the devel branch so will close this issue.

closed 2015-05-10 by mosh1