OctoPrint/OctoPrint #769: SerialExceptions after moving from 1.1.1-27 to 1.1.1-30

shanselman, 2015-02-15:
I recently updated from 1.1.1-27-g2523cf8 to 1.1.1-30-g4fede5a (master branch). 

While 1.1.1-27 was *rock solid* and I successfully printed a number of 8-12 hour long prints, now I'm seeing SerialExceptions, some a few hours in, others almost immediately. Here's the logs. I think something changed in comm.py, and I'm looking through it now.

(Trying to see how to downgrade also)

    Recv: T:136.6 E:0 W:?
    Unexpected error while reading serial port: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:982
    Changing monitoring state from 'Printing' to 'Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:982'
    Connection closed, closing down monitor

shanselman, 2015-02-15:
I'm switching to  1.2.0-dev-79-g2523cf8 to see if there's any issues there.

shanselman, 2015-02-15:
Here's the console log, to be expected

    ot event MetadataAnalysisStarted with payload: {"type":"gcode","file":"atx_back_top-try_2.gco"}
    dataupdater.js:168 Got event ClientOpened with payload: {"remoteAddress":"10.71.1.44"}
    dataupdater.js:168 Got event ClientOpened with payload: {"remoteAddress":"10.71.1.44"}
    dataupdater.js:168 Got event ClientClosed with payload: {"remoteAddress":"10.71.1.44"}
    dataupdater.js:168 Got event ClientClosed with payload: {"remoteAddress":"10.71.1.44"}
    dataupdater.js:168 Got event ClientOpened with payload: {"remoteAddress":"10.71.1.44"}
    dataupdater.js:168 Got event ClientOpened with payload: {"remoteAddress":"10.71.1.44"}
    jquery.min.js:6 POST http://octopi.local/api/connection 400 (BAD REQUEST)
    jquery.min.js:6 POST http://octopi.local/api/connection 400 (BAD REQUEST)
    dataupdater.js:168 Got event Disconnected with payload: null
    dataupdater.js:168 Got event Disconnected with payload: null
    dataupdater.js:168 Got event Error with payload: {"error":"Failed to open serial port, permissions     correct?"}
    renderer.js:345 Drawing layer 0 from 0 to 2339 (current: true)

shanselman, 2015-02-15:
OK, ignore me. Sorry to clutter this space, but hopefully this issue will be Googleable for those who see the same issue. The USB cable was shaking as the longer prints continued and it was causing issues.

closed 2015-02-15 by shanselman

HedronUser, 2015-05-15:
Having this same error when using octoprint, I've tried multiple cables, different electronics board, I'm stumped.

shanselman, 2015-05-15:
For me this ended up being that the cable was getting shaken by the printer
itself and wasn't properly seated.

HedronUser, 2015-05-15:
I just attached a different power supply and have had no problems after an 8 hour print...

HedronUser, 2015-05-18:
Well my second 8 hour print resulted in another of the same glitch error. I have not had a failure when I print from the micro SD card so I'm guessing it's a USB serial communication error. For reference, I'm using melzi ardentissimo electronics board from reprap.me with Repetier Firmware. For helpfulness, I'm using the 'M23 xxxx.gcode' and 'M24' commands to select file and start printing from SD, respectively.

th0ma5w, 2015-10-03:
I was experiencing this issue consistently with an iMac that I had converted to Ubuntu. With the same cable, I did not experience these errors. 

After much googling about these errors, I am convinced that it is one of several combined issues. I believe that in the cabling isn't loose like previous comments offer, that it could be either the USB controller, radio interference, subtle kernel level timing issues, or something mutable about the callstack involved in PySerial's readline() function. Essentially, Python's builtin select can show a state where the serial port is able to be read, but no data is available when it is subsequently read (just like the error states).

Pasted below is a patch for comm.py ... As you can see, this is a fairly basic kludge whereby instead of relying upon the PySerial readline() function's use of the "select" library (and hence the kernel select function) to infer the read buffer state, we actually make sure there is something in the buffer to be read, and only read a character at a time until a newline character is received, at which point we return the line. 

This made my completely unusable Octoprint install (couldn't even connect) able to print many things using a Micro M3D and the M3D-Fio plugin.

Hopefully this helps someone else. I'm not entirely sure this is desirable to be included into the project as ideally readline() should work, and this seems to be a rare issue. I'm not sure of the performance implications, or any other possible side effects of using this hack, but the PySerial implementation more or less works just like this except for the avoidance of the "select" library. 

--- ./comm_master.py	2015-10-03 13:31:37.066627060 -0400
+++ ./comm.py	2015-10-03 13:29:31.090629316 -0400
@@ -1342,7 +1342,15 @@
 			return None
 
 		try:
-			ret = self._serial.readline()
+			#ret = self._serial.readline()
+			if self._serial.inWaiting():
+				ret = ''
+				ret = self._serial.read(1)
+				while ret[-1] != '\n':
+					if self._serial.inWaiting():
+						ret += self._serial.read(1)
+			else:
+				ret = ''
 		except:
 			if not self._connection_closing:
 				self._logger.exception("Unexpected error while reading from serial port")

th0ma5w, 2015-10-09:
Just another update to this. After having all kinds of issues with the above hack, I undid this, and then I changed the equivalent of the lines in PySerial 2.7 in the serialposix.py in the spot roughly as annotated by this link https://github.com/pyserial/pyserial/blob/last-svn-state/pyserial/serial/serialposix.py#L469-L487 which is a newer version of the file. In effect, the first exception is commented out completely, the portNotOpenError, and then the last line, the SerialException one, I commented out and replaced with a "pass" statement. Octoprint is now stock at the master version, and everything works a lot better.