OctoPrint/OctoPrint #771: [Request] Add timestamp in terminal

fotis400, 2015-02-16:
During some debugging of my printer, I found that I had to mark down time while sending some commands and receiving the responses.

It would be of great help if we had a timestamp next to each line in the terminal window. Maybe it could be toggleable as the filters are.

cmmh, 2015-02-27:
Agree, would be helpful

mazamorac, 2020-05-05:
I'm calibrating thermistors by taking down the ADC values in the terminal (by turning on SHOW_TEMP_ADC_VALUES in Marlin). I'm comparing it to another known good thermistor rigged to a datalogger. If I had timestamps in the terminal, it would be trivial to match the data, and would result in much-increased accuracy, and save me about four hours of staring at two screens.

Pretty please with sugar on top? Thanks!

dryphi, 2020-12-10:
Any updates on this? I agree a timestamp would be useful. Could be checked on / off as needed.

aadeelnrx, 2022-09-24:
Hi, the serial log contains timestamps. 
To quote the serial.log's contents when it's disabled:
"You can enable it via Settings > Serial Connection > Log communication to serial.log"
Example output:
2022-09-24 17:11:48,661 - Recv: T:20.2 /0.0 B:20.1 /0.0 T0:20.2 /0.0 @:0 B@:0 P:21.0 A:27.0
2022-09-24 17:11:50,778 - Recv: T:20.6 /0.0 B:21.1 /0.0 T0:20.6 /0.0 @:0 B@:0 P:21.2 A:26.9
2022-09-24 17:11:52,889 - Recv: T:21.2 /0.0 B:20.4 /0.0 T0:21.2 /0.0 @:0 B@:0 P:21.3 A:27.1