OctoPrint/OctoPrint #772: Wrong parsing of Marlin temperature responce

fotis400, 2015-02-16:
There is an annoying bug long now, but for some reason I didn't opened an issue... So there you go.

When using the command `M303` in Marlin firmware the printer starts the PID autotune procedure, during which it responds with the current temperature, in the form: `Recv: ok T:151.09 @:0`.

The problem. When I execute `M303 E0`, regarding the hotend, the temperature graph updates correctly showing the current temperature of the nozzle. However when I execute `M303 E-1`, regarding the bed, octoprint does not parse the response and the graph remains freezed. 

Interestingly enough the `Suppress M105 requests/responses` works only when autotuning the hotend, while NOT while tuning the bed.

1. What were you doing?
Sending a M303 E-1 command to my printer.

2. What did you expect to happen?
See the temperature in the graph to follow the actual temperature.

3. What happened instead?
The graph remained freezed.

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-486-g7de90e1 (devel branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
Marlin with RAMPS (this issue persists for many months now, and for many Marlin and Octoprint versions)

I love cookies while watching my temperature graph...

fotis400, 2015-02-16:
You can also have a look at the temperature graph, after I've autotuned the hotend, and JUST after autotuning the bed.

Please take a look at the time axis and the labels....
[image]

foosel, 2015-02-17:
Hi @fotis400, can you please provide a serial log? I need to see what OctoPrint is receiving here that makes it go bananas.

Also make sure to take another look at [the contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md), they've been updated since you last read them (which the GitIssueBot should actually have told you about, but apparently that stupid thing still has a bug).

fotis400, 2015-02-17:
Ok sorry,  I didn't notice the changes in guidelines....

Here is the serial log. First autotune of extruder on line 6(it states "failed" but it is normal, it is my printer's...), and then autotune of the bed, starting on line 41.

I currently have only one extruder... Maybe the problem exists for multiple extruders also, I can't test it...

foosel, 2015-05-09:
Can you please not remove logs on still open tickets? At least pastebin says it was removed.

fotis400, 2015-05-15:
I am sorry for this, but I didn't remove it! I am also quite sure that I chose it to never expire... 

Unfortunately, currently I cannot recreate the log... If you really need it before I manage to repost it, please send me a message again.

foosel, 2016-01-25:
@fotis400 any news on this one?

fotis400, 2016-01-31:
Just tested it. As it seems it is now working fine. Some of the updates fixed it, and I didn't notice it.

You may close the issue.

foosel, 2016-01-31:
Can't remember touching the temperature parsing, but who knows... Closing.

closed 2016-01-31 by foosel

fotis400, 2016-01-31:
(Maybe Marlin changed the output format?)