OctoPrint/OctoPrint #774: [Question] Is Atcommand Plugin Still Supported?

eboston, 2015-02-17:
I posted a crash log that occurred with Atcommand a week ago but not seen any response to it.  Is the author of this plugin still supporting it?

foosel, 2015-02-22:
Sorry, I can't support third party plugins.

closed 2015-02-22 by foosel

eboston, 2015-02-22:
I wasn't asking for support. Just wanting to see if anyone knows if the author of the plugin, that you link too as a supported plugin, was still supporting it. If they aren't, you might consider removing the link.

foosel, 2015-02-22:
The **Bug**tracker is still the wrong location for something like this. Also: no, I'm personally not linking to anything or declaring anything as supported. The plugin list is part of a wiki. Everyone with a github account can edit it.

eboston, 2015-02-22:
I'm sorry.  I guess I didn't understand that for you, Issue means Bug.  While it wasn't directly related to OctoPrint, I thought this would be a good way to find someone who might know something since most of the people involved follow it.  I'm sorry for cluttering up your Bugtracker in trying to resolve an issue I was having with a plugin that is suppose to work with your project.

foosel, 2015-02-22:
> I guess I didn't understand that for you, Issue means Bug

This is actually explained in the guidelines linked to by GitIssueBot above. In those guidelines you also find links to better places for getting help with problems like these, to help in this exact kind of situations. 

Please understand that those guidelines are in place not to annoy people but to help keep the project manageable.