OctoPrint/OctoPrint #777: [Request] add speed & flow multiply option

MLWORKX, 2015-02-20:
- Sometimes it would be really helpfull to have a speed & flow multiply option as you have on RepetierHost (see picture).
- as I have a dual head printer it would be nice to have a seperate flow multiply for extruder 0 and 1

(currently using 1.2.0-dev-480-geb4156c (devel branch))

[image]

fotis400, 2015-02-21:
You can have this functionality by setting up some custom buttons.

Place in your config.yaml file the following:

```
controls:
- children:
  - command: M220 S%(percent)s
    input:
    - default: 100
      name: "Speed factor (%)"
      parameter: percent
    name: Set
    type: parametric_command
  - command: M221 S%(percent)s
    input:
    - default: 100
      name: "Extrude factor (%)"
      parameter: percent
    name: Set
    type: parametric_command
  name: Speed & Flow overrides
  type: section
```

MLWORKX, 2015-02-21:
That will work, thank you very much!
...but I think this should be more a "standard" feature which belongs into the "control" tab

foosel, 2015-02-22:
[x] you have not used the search function

Duplicate of #362

closed 2015-02-22 by foosel