OctoPrint/OctoPrint #778: disk full - URB error -71

sherifeid, 2015-02-21:
Hey Guys,

Ran the latest version of octoprint (1.1.1-30-g4fede5a) and for some reason the disk space gets filled up with kernel.log and syslog (in /var/log), I 1.4GB worth of logs repeating the same line

FFeb 20 17:52:46 octopi kernel: [90248.196751] gspca_zc3xx 1-1.5:1.0: URB error -71, resubmittinggspca_zc3xx 1-1.5:1.0: URB error -71, resubmitting
eb 20 17:52:46 octopi kernel: [90248.196751] gspca_zc3xx 1-1.5:1.0: URB error -71, resubmittinggspca_zc3xx 1-1.5:1.0: URB error -71, resubmitting

when googling I found out that it has to do with USB driver connections, the printing is functional, I can upload and print stls until the disk space hits the limit

at the moment I'm turning off kernel and syslog logging

foosel, 2015-02-21:
This has nothing to do with OctoPrint.

closed 2015-02-21 by foosel

foosel, 2015-02-21:
To give a bit more insight: This here is the issue tracker of OctoPrint, which is the platform agnostic print server serving the web interface ad controlling the printer. What you are experiencing is some issue your underlying Linux system is experiencing with your hardware, which is most probably caused by your hardware and not the underlying [OctoPi](https://github.com/guysoft/OctoPi) distribution. Your best bet to solve this problem is try to figure out what this gcpsa_zc3xx kernel driver is for and go from there to getting the piece of hardware plugged into your pi to work. A first google search reveals that it is a webcam driver, so try unplugging that, see if that makes the error go away, then try to fix it by more googling.