OctoPrint/OctoPrint #779: Restarting paused print retracts filament

CryoArchive, 2015-02-21:
I am running octopi with my printrbot jr v2. Haven't had any issues until now, where the filament is retracted all the way upon starting a print from pause. I would prefer this not happen. Is there a way to prevent this from happening in the future or is there a way to troubleshoot this, because this wasn't happening before. Thank you.

maribeiro, 2015-05-31:
This is happening to me aswell. When I pause a print, and change filament, the click 'pause' button agian, the head moves to where it paused but then starts retracting filament and filament gets out of the extruder.

foosel, 2015-05-31:
I. Need. Information.

You wouldn't just leave your car at a mechanic with the words "it stops driving" and then expect him to be able to solve the problem without further information than that, right?

https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report

maribeiro, 2015-06-01:
Correct!

I was running a print, and needed to change filament before layer height = 5, 
so I paused the print (via pause button on octopi UI) right before it changed to layer height 5.
After pausing the print, i moved the head X to 0 and did retract 100mm (to take out the filament) and inserted new one (and clicked extrude 100mm , filament started getting out), all ok!
After that, I hit 'paused' button again and the head moved to the last position (where I paused the print), as expected, but started retracting filament and the filament got out and it started printing without filament.
I was expecting that after inserting the new filament and hitting 'pause' button, the print would resume and there's no need to 'retract filament' after hitting 'pause' (for resuming print).

OctoPi Version: 1.1.1-30-g4fede5a (master branch)
Prusa i3 Hephestos by BQ ; Marlin development branch firmware

Google Chrome latest version 
Windows 8.1

foosel, 2015-06-01:
Please also provide a serial log of the problem.

Salandora, 2015-06-01:
Does this also happen when you use another host software?
I guess thus could also be a firmware bug.
That's why I ask.

markwal, 2015-06-01:
My guess would be that the E axis counter is affected by what is done during pause.  Then on continue the next gcode will have an E axis position that is less than current position and the firmware just does what its told and retracts to that E position.  The kicker is by what mechansim is the 100mm forward and backward done and is there any messing with the current position or switching to relative over absolute happening?

foosel, 2015-06-01:
@markwal @Salandora this is exactly why I'm asking for a `serial.log` here. From how I interpret the description above, I'm with markwal and it's running into some positioning issues (which is one of the reasons I want to look into position persisting when pausing). The retract/extruder commands on OctoPrint's UI should do a switch to relative positioning and then switch back to absolute, but maybe there's still a bug. In any case, all we can do is taking wild guesses here without seeing what actually goes over the line, hence the log.

nophead, 2015-06-01:
I don't think changing to relative mode and back again is sufficient. It only affects how the jog commands change the position of E but the problem is if the file is in absolute mode you need to restore the position of E without it actually moving.

maribeiro, 2015-06-02:
I found that the exactly same issue is happening on another firmware (craftbot firmware) 
see this post http://www.craftunique.com/forums/view-thread/629
so I guess this is related to firmware (however I use marlin not this craftbot firmware) ?!
I will try to replay the issue tonight and share the serial log

nophead, 2015-06-02:
Switch your slicer to relative E if you want to mess with the extruder during a print. It also makes cutting and pasting gcode practical.

maribeiro, 2015-06-02:
Cant find that option on Cura neither Slic3r. Any hint ?

nophead, 2015-06-02:
For Slic3r it is shown here: https://github.com/alexrj/Slic3r/wiki/Online-Help#printer-and-filament
I don't think Cura supports it: https://github.com/daid/Cura/issues/514

markwal, 2015-06-02:
In Slic3r it's on the Print Settings tab

https://github.com/foosel/OctoPrint/issues/779#issuecomment-108100159

maribeiro, 2015-06-02:
I cannot get slic3r to work as good as Cura does... so is there any other alternative besides switching the slicer ?

markwal, 2015-06-02:
Well, you could give your serial.log so we can see if there is a fix we can make to work with what your firmware is doing.

nophead, 2015-06-02:
Yes you need to set the correct mode in your start gcode.

For it to work with absolute E I think you would need to read the current E position with M114 set it to relative, jog the extruder, set it back to absolute and then restore the old E value with G92.

maribeiro, 2015-06-02:
Ok, it sounds better if I provide the serial.log... give me just a minute

maribeiro, 2015-06-02:
Here it is: http://pastebin.com/JfsBbkv2
So, i started printing, waited like 20seconds, clicked 'pause' and then moved the head (X axis) to 0 (using the octopi Control interface) , then retracted 80mm , extruded 80 but filament failed to enter the extruder, so I extruded another 3 times, filament started to get out... all OK ! 
Then clicked pause (to resume print) and the first thing it did was pop all the filament out (RETRACTING, it seemed like 80mm at least, may be more!) :/ then it moved the head to where it had paused but printing with nothing, since there was no filament because it retracted it all.

nophead, 2015-06-03:
Yes with absolute it will simply rewind to the position it was in before you moved it. The sum of your moves was -80 +80 +80 + 80 = 160 so it will go backwards 160 as soon as it sees the next E move in the gcode.

maribeiro, 2015-06-03:
So, what's the simpler way of solving this ?

nophead, 2015-06-03:
Use relative E mode and persuade @daid to implement it in Cura.
 
The more complex solution is for OctoPrint to read the E value with M114 when you pause and restore it with G92 when it restarts.

markwal, 2015-06-03:
As a workaround in the mean time you could do it yourself with the terminal window. When you hit pause, write down the E value from the last command the bot got.  After you're done with the reload, type a G92 command into the terminal window that looks like this:
```
G92 Enn.nn
```
I haven't tried it, but it should work.

The other way to go is to insert the pause directly into the G-code.  There's a plugin for Cura that does pause at Z.  Since it pauses at a known position, you can make sure to set that position at G92 on resume.  Some slicers reset E to 0 on every layer change even in absolute mode because it helps with loss of resolution of the value depending on how the math is done.

foosel, 2015-06-03:
I was looking the other day into reading the current position on pausing
and while it would be possible to do that, it would solve the problem only
for the first extruder in a multi extruder setup, since M114 only reports
the position of that one. Hooray.

daid, 2015-06-03:
Problem is more "getting the current active extruder" I think.

M114 -> store E

G92 E0
G1 E100 F900 <- move material

G92 E[stored_e_value]

Should work, even in dual extrusion, except when you also switch extruders when moving the material in a pause.

foosel, 2015-06-03:
> except when you also switch extruders when moving the material in a pause.

This is exactly my point. Might be possible to track the current tool and switching back to that before resuming and performing the `G92` but it's still somewhat ugly, especially since it wouldn't work for printing from SD card. In any case, nothing for v1.2.0.

Marking this as a Feature Request since it's not a bug, just "unexpected behaviour" of the printer from the OP's perspective for which it would be nice to have a helping hand in OctoPrint.

bolsoncerrado, 2015-12-17:
This is exactly happening to me as well using 1.2.8

I just paused a job after filament ran out, put in new filament, resumedn job and I could clearly see how filament was spit out of the extruder under a fast Retract motion coming from Octoprint :??????

bolsoncerrado, 2015-12-17:
Just noticed this only happens if you use the EXTRUDE function while the print is PAUSED. Then when you resume it, octoprint RETRACTS all the amount you extruded while in pause....

nophead, 2015-12-17:
It will if you use absolute E. You need relative E if you want play with
the extruder during the print, or hand modify g-code such as splicing files
together.

crprintoctopi25, 2016-02-12:
Hello,

I am having this happening on v. 
- OctoPrint: 1.2.9 
- Installed: 1.2.9
- Available: 1.2.9 (stable)
- Release Notes: https://github.com/foosel/OctoPrint/releases/tag/1.2.9

Would we then need to manually extrude to pre-prime the nozzle prior to resuming from a pause thru Gcode? (is this working behind the pause command's back or is it the same as hitting the extrude button?)

If so, please let me know suggested steps.  I had to change filament because of a miscalculated print / roll left.

In cura, you pause, move around, extrude-prime, then resume brings it right back to where it was.

Asking someone to insert a pause into the gcode is great, when you know you need to pause for some planned reason of course.

So, unless I'm missing something, because I'm new to octoprint, is there a button to "prime" say vs extrude when paused?  If not, does it need to be done thru gcode because of different printers (makes)/firmware/slicing software???

Also, dumb question, but can you pause a print,
say move head away so it doesn't melt into print, then turn off extruder / bed temps if you wanted to pause and not have all heated up?  Then after x amt. of time, can you re-heat, then un-pause or will things time out???

Thanks for your help!

averter, 2016-08-27:
+1 from me as well.

foosel, 2017-01-27:
1.3.0 will track the position during a pause, including the E coordinate, via `M114` and allow to utilize that data in the `onBeforeResume` GCODE script for more intelligent pausing: http://docs.octoprint.org/en/master/features/gcode_scripts.html#context

See also the [changelog of 1.3.0](https://github.com/foosel/OctoPrint/releases/tag/1.3.0) which contains [this example](https://gist.github.com/foosel/1c09e269b1c0bb7a471c20eef50c8d3e).

Note that due to the aforementioned issue with firmware not reporting back *all* E values for multi extruder setups but only the current one, this example will not work as flawlessly in such cases since OctoPrint simply lacks the information about any but the current extruder. Additionally there's no (generally applicable) way to request the current feedrate or selected extruder from the firmware either. OctoPrint tries to track that based on the sent data, but that of course doesn't work when printing directly from the printer's SD card. These limitations are also the reason why OctoPrint isn't shipped with the adjusted resume code but relies on the user to adjust it.

This is as good as it gets, thanks to firmware not sharing all information required to properly solve this (as is sadly way too common), so I'm closing this now as implemented.

closed 2017-01-27 by foosel

mihaisecasiu, 2018-02-18:
I believe a version of this is still happening in OctoPrint 1.3.6. If I cancel a print and then I try to print again by clicking on the "print" button in the list of files, it first retracts all the filament and then starts printing without filament. To fix this I have to restart octoprint.

markwal, 2018-02-18:
Yes, this will happen. A likely case is if you've got a GPX printer and you've got some old start gcode designed for single shot GPX, for example.

mihaisecasiu, 2018-02-18:
I'm using a Tevo Little Monster with the following start code: 
M280 S10.6
G28
M280 S3
G1 Z50 F25000
G0 Z10
M280 S7
G1 X0 Y0 Z100 ; lift nozzle
M42

Nothing in the above code suggests that it should retract when it starts. And why does it retract only after a cancel ?

Is there some code that I should use in the "after cancel" code to prevent this?

markwal, 2018-02-18:
Yes. The above code does not reset the extruder position, so a retract is expected. Perhaps your end gcode resets the extruder position. You could do it in the after cancel gcode, but it'd be more reliable in the start gcode.

On other printers it would be G92 E0, not sure on your printer.

c3r63ru5, 2018-06-18:
I know this is old and closed, just an additional hint for users experiencing the filament retraction after resume: Check your slicer settings, I was using slic3r which is using relative extruder moder by default. So, to make this work you need to use the relevant GCODE (Octoprint Docs) and remove the switching back to absolute mode (and I guess the extruder reset).