OctoPrint/OctoPrint #780: [BUG?] Cant resume printing

bolsoncerrado, 2015-02-21:
There are times I cant click the PRINT button on the file manager unless I have another file in the SD card... It happened I installed OctoPi from scratch then upload one file and upon cancelling and willing to reprint it, i couldnt unless I uploaded a second file, then select it, then i could select the print again...

Hard to explain, I know... May be a video?

nophead, 2015-02-22:
Yes that is always the case for me. I can never print the same thing again
without loading something else. All versions I have used, currently
Version: 1.1.1-22-g83790fc (master branch).

Changing monitoring state from 'Operational' to 'Printing from SD'
Send: M24
Recv: ok
Send: M105
Send: M27
Recv: ok T:19.7 /0.0 B:19.9 /0.0 T0:19.7 /0.0 @:0.00 B@:0.00
Recv: SD printing byte 2760322/2760322
Recv: ok

Nothing happens and the SD index stays at the end of the file. In
Marlin M24 just does sets
sdprinting = true;. It doesn't reset the file pointer.

foosel, 2015-02-22:
Thank you Chris. That amount of information in fact helps solving problems like these.

commit 5c35a288 (2015-02-22): Always make sure to (re)set file position to the beginning when starting SD card prints

Closes #780
commit 89d972ff (2015-02-22): Always make sure to (re)set file position to the beginning when starting SD card prints

Closes #780
(cherry picked from commit 5c35a28)
closed 2015-02-22 by foosel