OctoPrint/OctoPrint #783: [Request] event user logged in

n-dim, 2015-02-24:
The event ```ClientOpened``` with the payload if its IP address exists already, but I think an event for ```user logged in``` with the payload  ```username``` could be useful.

foosel, 2016-06-09:
I was just about to finally add this when I realized that it might be a really bad idea considering that events are public (pushed to any connected client regardless of login state, since the server doesn't know that for socket clients) and hence such a event with corresponding payload could be used to prepare an attack by sniffing out possible usernames to use. And that would not be a good idea.

So in order to allow realization of this, there needs to be a way to distinguish between logged in socket sessions and anonymous socket sessions and a limitation of events (or possibly event payload) with regard to permissions associated with the session (also a point to consider for #1110).

And while tackling that it might also make a lot of sense to strip the IP from the event payload of the `ClientOpened` event for anonymous users.

foosel, 2019-10-29:
Implemented by the above commit, will be part of 1.4.0

foosel, 2020-03-04:
1.4.0 has been released.

closed 2020-03-04 by foosel