OctoPrint/OctoPrint #787: File list icons do not work behind https reverse proxy

ptillemans, 2015-02-26:
Hi,

I am running Octoprint on a Raspberry Pi behind a haproxy which accepts connections on https and proxies it to the Octoprint backend.

I noticed that I could not load new gcode files anymore after upgrading to the head of the master branch (commit d974ab0).

When clicking on the load icon, nothing happens. Same thing for load+print or delete for that matter.

In the javascript console I see that the request is blocked because it is doing XmlHTTPRequest to a unsecured http port from a secured https page.

A workaround for me was to connect directly to the 5000 port over http.

octoprint log: https://gist.github.com/ptillemans/a730a6158d59398ad29c

Javascript console: 
Laden van gemengde actieve inhoud “http://192.168.3.32/api/files/local/Fuel_cell_car_tank_v1_3.gcode” geblokkeerd[Meer info] jquery.min.js:6

[image]

I have read the FAQ.

foosel, 2015-02-26:
Please verify your haproxy configuration correctly sets the X-Scheme header as described [here](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples#haproxy) in case of SSL connections, otherwise OctoPrint won't know that it has to generate ``https`` URLs.

ptillemans, 2015-02-26:
Thanks. I misclassified it as a bug because this worked for months and the only thing that had changed was upgrading to the latest and greatest and the other links essentially worked. 

Thanks for the awesome product and the lightning fast response.

closed 2015-02-26 by ptillemans

foosel, 2015-02-26:
No problem, I'm always glad when it's something simple as this. Also added this problem to the FAQ now since you are not the first person to run into it. Cheers!