OctoPrint/OctoPrint #795: Can't Load Cura Profile

nacho693, 2015-03-02:
1. What were you doing?
Attempting to load Cura profile.
2. What did you expect to happen?
Successfully load profile.
3. What happened instead?
Interface stalls and does nothing. Window stays open.
4. Branch & Commit or Version of OctoPrint:
    Version: 1.2.0-dev-563-gdec9f47 (devel branch
5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
MakerGear M2
6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Firefox 36.0
7. Link to octoprint.log on gist.github.com or pastebin.com:
http://pastebin.com/itZNaJqk
8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure):
N/A
9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure):
http://pastebin.com/zuEuVCG0
10. Screenshot(s) showing the problem (if applicable - always
    include if unsure):
http://snag.gy/tq4d0.jpg
I have read the FAQ.

foosel, 2015-03-03:
Should be fixed in ``devel``

nacho693, 2015-03-03:
Same issue but no log entries. Still reporting in console:

Empty string passed to getElementById(). jquery.min.js:3:0

foosel, 2015-03-03:
Have you cleared your browser's cache? If so, please provide the stack trace for that error.

nacho693, 2015-03-03:
Yes, cleared cache.  Never done a stack trace so downloaded firebug extension and called "trace" on the error.  Does http://pastebin.com/LCrbKZsw help?

Also, I noticed a behavior that if I select the .ini for upload and *don't* fill in the fields to override the default input text, most of the time it will upload the file.

foosel, 2015-03-03:
Hm, that trace sadly doesn't help. And I cannot reproduce it at all, current version in firefox works fine, just as expected. Do you have any additional errors in your console that might indicate that it's not fully loading the files? It sounds like it can't properly attach the view model to the dialog due to that error (that's just a wild guess though, since I can only guess without a proper trace to work on), but from the code there's definitely no empty string passed here. What you could do is in your browser download ``http://<your octoprint url>/plugin_assets/cura/js/cura.js``, see if that ends with:

```
    // view model class, parameters for constructor, container to bind to
    OCTOPRINT_VIEWMODELS.push([
        CuraViewModel,
        ["loginStateViewModel", "settingsViewModel", "slicingViewModel"],
        "#settings_plugin_cura"
    ]);
```

Please also try another browser (e.g. Chrome) to see if that error persists there (if so, that would at least give us a proper stack trace, if not it would narrow down the issue).

nacho693, 2015-03-03:
Reproduced on Chrome. Here's from its JS console:

http://pastebin.com/fXrV35E7

Thank you

nacho693, 2015-03-03:
OH and yes, the cura.js tail ends fine.  Thank you

foosel, 2015-03-03:
That looks like a different error now than before though. Next step, can you please open the network tab in Chrome's web development tools, find that POST request that's returning the 400 Bad Request there and take a look at it's response by clicking on it? E.g.:

[image]

[image]

nacho693, 2015-03-04:
Yes, "Could not convert Cura profile".

Thank you.

kev626, 2015-03-05:
I am having this same problem too...

foosel, 2015-03-05:
Both of you, please provide the profiles that fail to import. Apparently something's in them that OctoPrint can't parse and I'll need examples to figure out what that is.

nacho693, 2015-03-05:
https://drive.google.com/file/d/0BwhomOh6F9rSNmRpQXlQV2JZVHM/view?usp=sharing

It's from https://github.com/kefir-/M2-config/

As previously mentioned, I can import it if I don't customize the default name/description inputs.

foosel, 2015-03-05:
When you are customizing it, what are you putting in instead of the
defaults?

nacho693, 2015-03-05:
Various attempts but most recently: asdf in all three fields.  Have tried
with overwrite on and off.

Thanks

foosel, 2015-03-23:
Hm... I just tried the profile (sorry it took so long), can import it just fine, with and without customization. Any others that don't work? I cannot reproduce it at all :/

nacho693, 2015-03-24:
I'll try some others when I get some time.  The devel install went on the
backburner.  Thanks for your effort.

closed 2015-06-10 by foosel

willyt, 2016-12-07:
I'm still seeing this also on mac safari and mac firefox with Version: 1.2.10 and Version: 1.2.18 The ini file I'm trying to import is attached. Hitting the Confirm button does nothing.
[cura.txt](https://github.com/foosel/OctoPrint/files/637202/cura.txt)
How can I help to troubleshoot this?

foosel, 2016-12-08:
@willyt That doesn't look like a Cura 15.x profile. [This is what a Cura 15.x profile looks like](https://gist.github.com/foosel/b6196e7fe4aab1f8f62c6ee39ac2b920). 

Your's is missing the headers, and the keys are also named completely differently. Where's that profile from? The Cura plugin currently bundled with OctoPrint supports only profile `.ini` files exported from Cura versions up to and including 15.04.06. Anything from newer Cura versions (which - careful - are named Cura 2.x) is not supported. Neither are profiles exported from third party tools such as Repetier Host.

See [OctoPrint On Air Episode 4](https://www.youtube.com/watch?v=VsJ2DmbIwzs&feature=youtu.be&t=28m40s) for a short demonstration of how to get such a profile.

johnpettybone, 2018-01-12:
I had this same problem, and then noticed the "Confirm" button still spinning to show "busy" even after I aborted. I closed the instance of Octoprint and started a new one. This solved problem.

agribov, 2018-01-15:
@johnpettybone -- What procedure exactly did you use to restart your instance? I have the same issue (made a new thread, #2368 ) and haven't been able to resolve it by restarting Octoprint through the browser.

johnpettybone, 2018-01-15:
@agribov I tried deleting the cache in Chrome first, but this didnt stop the “busy” animation spinning on the “confirm” button on the “add profile” pop up page. The next thing I did was go to the “Instance” drop down menu at the top left of the Octoprint GUI, and chose to close the current instance. Once Octo reloaded the animation on the confirm button was gone, and I was able to add as many profiles as I wanted. Clearing the cache in conjunction may be necessary. Just closing the browser wouldnt resolve it for me. I had tried that several times. It only stopped the program from resuming right back into the loop when I used the drop down box to “close this instance of Octoprint”. Hope this helps. I am not a programmer, so I cant technically tell you what is happening script wise.

mrenouf, 2018-06-11:
Still broken for OctoPi install of 1.3.8.
Cannot import cura profile for Prusa i3 Mk3

[Original Prusa i3 MK3.ini.txt](https://github.com/foosel/OctoPrint/files/2090392/Original.Prusa.i3.MK3.ini.txt)

foosel, 2018-06-12:
@mrenouf 

> `# generated by Slic3r Prusa Edition 1.38.6-prusa3d-win64 on 2018-02-22 at 22:24:03`

Yeah, I'm not surprised that the **Cura** plugin can't import a **Slic3r** profile ;)

commit befe2e7b (2015-03-02): Allow adding slicing profiles for unconfigured slicers and properly reload slicing view model upon configuration

This should fix the problem where on a fresh setup it was impossible to upload slicing profiles for Cura before
the path to the binary was configured.

Also made the slicing dialog auto update available slicers when the settings are updated. The slicing button
in the file list is now only active if a slicer is available. The slicing dialog will only show upon upload of
an STL file if a slicer is available.

Closes #795