OctoPrint/OctoPrint #796: G29 Command not working if printing from internal memory

jorgeiglesiasca, 2015-03-02:
I´m facing sice i started last week to use Octoprint with octopi distro.

I make my Gcode files with kisslices as i used to with repetier but once i upload them to the octoprint GUI if i upload to the octoprint server (PI SD card) it does the G28 command but not the G29 just after.

If i upload it to the SD on the printer thru the octoprint interface it takes ages , it does not give proper information about the time left and the amount of file printed but it does execute the G29 Auto-Level pattern when it should be.

Tried Master - Devel and Beta branches to see if it was a bug or not. but still happening. Its something i´m doing wrong or its a software fault.

What were you doing?
Sending G29 Code from Octoprint to printer prior to print just after G28

What did you expect to happen?
Auto - Level process

What happened instead?
Jumped to next Gcode on file doing nothing

Branch & Commit or Version of OctoPrint:
Master , Deve, and beta shows the same behaviour

Printer model & used firmware incl. version (if applicable - always include if unsure):
Octoprint (raspberry - pi) - P3Steel with Latest marlin and all the auto-level parameters defined and tested from diferent softwaare like repetier and cura with no problem on G29

Browser and Version of Browser, Operating System running Browser (if applicable - always include if unsure):
Windows 7 64 - Google Chrome 64 -

Link to octoprint.log on gist.github.com or pastebin.com:
http://pastebin.com/NfJG39Xv

Link to contents of terminal tab or serial.log on gist.github.com or pastebin.com (if applicable - always include if unsure): N/A

Link to contents of Javascript console in the browser on gist.github.com or pastebin.com or alternatively a screenshot (if applicable - always include if unsure): N/A

Screenshot(s) showing the problem (if applicable - always include if unsure): N/a

I have read the FAQ.

foosel, 2015-03-03:
> Link to contents of terminal tab or serial.log on gist.github.com or pastebin.com (if applicable - always include if unsure): N/A

I need this.

foosel, 2015-03-23:
No response, closing

closed 2015-03-23 by foosel