OctoPrint/OctoPrint #798: Octoprint + Rasberry Pi B + Melzi = NOT CONNECTING

jradler, 2015-03-05:
So I use the Melzi 3d printer controller board but for some reason my Octoprint will not connect nor will it show live camera but I can reboot and type code into the terminal and control the raspberry pi strictly but it wont control the 3d printer for some reason. PLEASE HELP

I have attached what it shows me on octoprint:
[image]

Thank you all for your help it is very appreciated !!!!

closed 2015-03-05 by GitIssueBot

nophead, 2015-03-05:
That is the wrong serial port. It should /dev/ttyUSB0 for a Melzi.

/dev/ttyAMA0 is the RPi serial port on the GPIO connector.

jradler, 2015-03-05:
I have tried using that port it did not work any other suggestions

foosel, 2015-03-05:
Read the FAQ

jradler, 2015-03-05:
I have if there is a fix for it that I guess I have not found please paste it in here

nophead, 2015-03-06:
Set the baud rate correctly to match your firmware and increase the timeouts to 20s if you have the auto reset jumper fitted.