OctoPrint/OctoPrint #800: Cura 15.02.01 always generate Error: Format error

tiagojanz, 2015-03-07:
1. What were you doing?
Try to print a gco file produced by Cura 15.02.01

2. What did you expect to happen
The file printed successfully

3. What happened instead?
Octoprint disconnects and send   Error: Format error
 

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-619-gdee205f (devel branch)

5. Printer model & used firmware incl. version
Prusa I3 / repetier

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Linux Ubuntu 14.10 / Chrome

7. Link to octoprint.log on gist.github.com or pastebin.com:
http://pastebin.com/tjuZnPb5

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues):
http://pastebin.com/tjuZnPb5

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

foosel, 2015-03-07:
I'll need the serial.log but your second link is octoprint.log too, please provide it.

tiagojanz, 2015-03-07:
I found out what was wrong!!
The start gcode  that comes with cura have this line

G1 Z15.0 F{max_z_speed} ;move the platform down 15mm

after saving the gco files it appears like

G1 Z15 F?Z_TRAVEL_SPEED?

[image: Inline image 1]

foosel, 2015-03-07:
Did you slice through OctoPrint's Cura integration, or through Cura directly and then upload the file to OctoPrint?

tiagojanz, 2015-03-07:
Cura directly and then uploaded to OctoPrint, Its dificult to use Cura
integration because you can't reorient the object, and I have a weird
problem that after it homes all axis the printhead goes to some negative
coordinates.

foosel, 2015-03-07:
In that case at least it's fairly certain that it's a bug in Cura, not OctoPrint. Closing this therefore. FYI, the reorientation and otherwise full featured manipulation (scaling, rotating, positioning) is on the TODO list, but not for 1.2.0, which I'm trying to finalize now (document, stabilize etc)

closed 2015-03-07 by foosel