OctoPrint/OctoPrint #804: [Request] Restart/Reboot/Shutdown through Api

mzander, 2015-03-09:
Hi there,

I am the developer of OctoDroid for Android.
An often requested feature is that the user wish to perform a reboot/shutdown/restart of OctoPrint through OctoDroid.

Are there any plans to make it available through the Api?
When I am currently calling the propper api call the api response says 'not allowed'

Best regards,
Moritz

foosel, 2015-03-09:
Should have left it at "Request", there's no such thing as "Question" anymore (compare [CONTRIBUTING.md](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md))

foosel, 2017-01-27:
Was always possible, just not documented. Since 1.3.0 it finally is documented too: http://docs.octoprint.org/en/master/api/system.html

closed 2017-01-27 by foosel