OctoPrint/OctoPrint #805: [Request] OctoPrint & GUI in Kiosk mode on Raspberry 2

scp38, 2015-03-09:
1. What were you doing?
I'm running Octoprint on a Raspberry Pi 2B (on Raspbian), together with an **instance of the Octoprint GUI in the kiosk mode of KWeb3**. I have mounted the new Raspberry 2B and a 7" TouchScreen Display to my printer to have a local Octoprint interface directly at the printer without the need to start a PC.

2. What did you expect to happen?
Octoprint GUI running nicely and responsive.
NO active GCode Viewer.

3. What happened instead?
Octoprint GUI in Kweb3 kiosk mode is sometimes not responsive and on some ocasions Kweb3 is crashing.
The GCode Viewer is active; the screen does NOT show the message that I'm running a weak machine... which I'm seeing on the Smartphone or the Tablet.
I wonder if it is a good idea to run the Octoprint Browser GUI instance locally on the same Raspi box, but I have the feeling that there isn't much missing with the new Raspi 2 and its Quad Core CPU; it works right out of the box. Maybe the task is just keep the local Browser instance off from jobs requiring heavy CPU load, e.g. displaying the GCode Viewer by default and/or the webcam stream.
Any ideas?

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.0-dev-619-gdee205f (devel branch)

5. Printer model & used firmware incl. version
Felix 3.0 duo

6. Browser and Version of Browser, Operating System running Browser
Kweb3 on Raspbian/Raspberry 2B

7. Link to octoprint.log on gist.github.com or pastebin.com:
n/a

8. Link to contents of terminal tab or serial.log on gist.github.com or pastebin.com 
n/a

9. Link to contents of Javascript console in the browser  on gist.github.com or pastebin.com or alternatively a screenshot 
n/a (let me know if and what may be of interest)

10. Screenshot(s) showing the problem 
n/a

I have read the FAQ.

scp38, 2015-03-31:
I made some investigations on the issue today:
- if the "Enable GCode Visualizer" is enabled, the GCode viewer is always showing the visualisation. The browser or machine (raspian/Raspberry) is NOT considered a mobile browser. That can lead to high CPU load and claiming of all memory, depending on the size of the GCode file
- The decision whether the GCode shall be visualized is taken locally in the browser (in this case kWeb3) in **gcode.js**

 ```javascript
if (data.job.file.size > CONFIG_GCODE_SIZE_THRESHOLD || 
			($.browser.mobile && data.job.file.size > CONFIG_GCODE_MOBILE_SIZE_THRESHOLD))
```
- The determination whether the javascript runs on a "mobile" browser is done in **detectmobilebrowser.js** 

However, I'm lost on how one could make javascript detect the local browser (here kweb3) and/or raspbian in a decent way.

foosel, 2015-04-01:
The only thing I can think of is detecting that specific user agent manually in gcode.js as well, ideally via another utility function. Something like 

``` js
if (data.job.file.size > CONFIG_GCODE_SIZE_THRESHOLD || 
        (($.browser.mobile || $.browser.embedded) && data.job.file.size > CONFIG_GCODE_MOBILE_SIZE_THRESHOLD))
```

and of course adding some kind of User Agent matching function as `$.browser.embedded` similar to what is going on in `$.browser.mobile`.

scp38, 2015-04-01:
Hmm, sounds reasonable to me. 
Having said that I have no experience with javascript at all, i managed to run a script to get the responses of the browsers on Raspbian to 
- navigator.appversion
- navigator.userAgent
- navigator.vendor

However, the result is rather astonishing...

**kweb3 (1.6.4):**
**appversion:** 5.0 (Macintosh; ARM Mac OS X) AppleWebKit/538.15 (KHTML, like Gecko) Safari/538.15 Version/6.0
**userAgent:** Mozilla/5.0 (Macintosh; ARM Mac OS X) AppleWebKit/538.15 (KHTML, like Gecko) Safari/538.15 Version/6.0
**vendor:** Apple Computer, Inc.

**Epiphany:**
**appversion:** 5.0 (Macintosh; ARM Mac OS X) AppleWebKit/538.15 (KHTML, like Gecko) Safari/538.15 Version/6.0 Debian/7.8 (3.8.2.0-0rpi18rpi1) Epiphany/3.8.2
**userAgent:** Mozilla/5.0 (Macintosh; ARM Mac OS X) AppleWebKit/538.15 (KHTML, like Gecko) Safari/538.15 Version/6.0 Debian/7.8 (3.8.2.0-0rpi18rpi1) Epiphany/3.8.2
**vendor:** Apple Computer, Inc.

For comparison: **Firefox on Win7:**
**appversion:** 5.0 (Windows)
**userAgent:** Mozilla/5.0 (Windows NT 6.1; WOW64; rv:36.0) Gecko/20100101 Firefox/36.0
**vendor:**

moritzmhmk, 2015-09-18:
Instead of using the normal web UI in kiosk mode I decided to write a custom one, which you might want to check out here [OctoPrintNano](https://github.com/moritzmhmk/OctoPrintNano)
[image]
[image]
[image]

foosel, 2017-01-27:
Duplicate of #215

closed 2017-01-27 by foosel