OctoPrint/OctoPrint #811: [Request] add plugin support

amedranogil, 2015-03-13:
The idea is to add capability for web hooks to be configured by an administrator, that are loaded when some action takes place (added new model, starting printing, finished printing, when taking a snapshot,...)

This way it is possible to easily create a plugin that, for example turns a light on before taking a snapshot, or sends an SMS when your print is done (email sending can be reworked as a plugin)

The web hooks should also provide important information; as for the main plugin I am interested is an accounting plugin that tracks the printer usage time, the amount of material used, ratio of successful prints vs cancelled prints, ... (if i may add another though for this plugin, using the logged information it can predict when the printer needs maintenance).

Web hooks may also be placed on octoprint side, so some actions are triggered when the hook is called (but almost sure octoprint API is all about these).

foosel, 2015-03-13:
You mean something like the [event hooks](http://docs.octoprint.org/en/master/events/index.html) which have been in the software since like two years?

amedranogil, 2015-03-16:
Yes, I was looking for this in the main features list, but the link was a bit hidden; I was highly surprised there wasn't any feature like this.

Although it is also true for true "pluging support" it would be also nice to have some sort of API to access UI ... which I guess it is the [custom controls feature] (https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls). 

Just skimming over the documentation for both features it seems to be enough to offer "plugin support", but the idea for "plugins" which use both events and UI customization in a single "package" seems not to be connected.

If the community starts developing more custom UIs, events (or plugins), will they inform the potential users by updating the cookbook? or is there anywhere a full list of these addons?

foosel, 2015-03-16:
That's at least the basic idea behind it and what's available in the current stable release.

However, the current `devel` version of OctoPrint (instructions on how to switch branches [can be found on the wiki](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image)) also has a proper plugin system which allows way more customization options than custom controls or event handlers so far did. Take a look at the [devel docs](http://docs.octoprint.org/en/devel/plugins/) (work in progress) for some insights in that regard.

I agree that the full list of all available features in the software is still hard to puzzle together. Rectifying this situation is actually quite high on my TODO list since I keep getting tickets like this one ;)

closed 2015-03-16 by foosel